espruino/BangleApps #555: [Feature request] Improve signal processing for step counter and HRM

singintime, 2020-08-31:
Working on the banglerun app, I find the step counter and HRM readings pretty inaccurate under heavy exercise conditions.
I propose the following enhancements to the DSP algorithms:

- step counter:
    - separate "movement" from "gravity" with a crossover filter
    - taking the dot product of movement and gravity, to detect movement in the direction of gravity
    - applying a threshold, either static (as currently done) or adaptive

- HRM:
    - option 1: using statistical moments (skewness and kurtosis) of the time series to detect and discard motion artifacts, then performing autocorrelation in the 40-200Hz range as it is done now
    - option 2: using accelerometer input to clean the HRM signal from motion artifacts, by doing spectrogram or correlogram subtraction

I would be willing to work on that, but I'd need some help and guidance to build the espruino firmware for bangle.js

gfwilliams, 2020-08-31:
Hi, I'm totally open to improving the algorithms if you feel you can improve them. Building Espruino itself should be pretty straightforward but ask on the forum if you have any issues.

> step counter

There's an issue open for this here: https://github.com/espruino/Espruino/issues/1846

There's already what looks like a good, proven step counter that we could pull in, which could be a good start?

> HRM

You're welcome to try, but I don't hold out much hope for this I'm afraid. If you look at the data that comes in you'll see that after movement the HRM hardware basically becomes saturated and doesn't give you *any* useful data for maybe 10s, so it's not a matter of it being obscured by movement - there's nothing there to extract :(

What could *really* be good is to sort out the confidence value though (maybe using the accelerometer so it didn't just trigger on repetitive movements), and at least then apps could run HRM until it looked like you got a clean value.

m0n5t3r, 2020-10-14:
I was playing with the heart rate monitor app recently while doing some breathing exercise, it seems almost useless when worn as a watch, but works perfectly when pressed against a finger; I'll dig out my old lenovo hx03w tracker to compare LED brightness if I can find it, because that one seemed pretty reliable and the sensor ended up roughly in the same position on my wrist. I'm guessing a hardware modification to change the current would be pretty tricky, though.

gfwilliams, 2020-10-15:
Interesting that HRM works better when your finger is on it. I honestly think it's a hardware issue though - I didn't mention above but:

The HRM value being useless 10 sec after movement is because the HRM is analog. There's a capacitor + OpAmp in there that tries to find the right level so the heart rate is picked up, but when you move the value changes so wildly that the level is knocked off and takes a while to readjust.

I guess there are 3 potential hardware hacks:

* Bigger capacitor value - takes longer to get an HRM value but less easily knocked off
* Smaller capacitor value - regains HRM faster after getting knocked off, but knocked off more easily
* Change some resistor values to make HRM less sensitive, then do more signal processing in software to extract the signal

But none of those is a great solution for most people as they'll require hardware mods

gfwilliams, 2021-12-02:
I think this is done now - there's a now algorithm for both since this post

closed 2021-12-02 by gfwilliams

m0n5t3r, 2021-12-03:
updated today, HRM is definitely fixed, but both pedometers (original and active) are just as useless as before :)