espruino/BangleApps #564: Bangle.js Randomly Crashes All The Time

Shashank-Prasanna, 2020-09-23:
Hello All, 
So here's my issue. I love the Bangle, I really do, but I can't stand it crashing all the time. I'm checking the time or looking at the calendar or playing Pong or whatever, and then suddenly **BOOM!** It resets, the time goes back to the default. And this isn't a rare occurrence, either. It happens a solid 5-10 times a day, and I use the Bangle for maybe 1/2-1 hour a day. I hope someone can tell me what is going on! I also have 2430548 memory left (I think the unit is bytes but am not sure), so I don't think it is a low memory crash. Please help!

gfwilliams, 2020-09-24:
That's a really odd one. Does it appear to hang for a few seconds before it reboots, or does it just reboot without warning? Are you using Gadgetbridge?

Have you found a way to force it to reboot so that we could reproduce it here?

Is your Bangle.js watch firmware up to date? 2v07 is the latest now - it'd be well worth updating.

The default watch firmware doesn't reboot at all, so realistically it's going to be related to some non-default setting, or some widget that's causing problems. Please could you screenshot your 'my apps' page of the app loader so we can see what's installed?

It'd be a huge help if you could do 'Install Default Apps' in the app loader (which will return the Bangle to more of a 'factory' state) and then re-add widgets one at a time until you see the errors start to occur - then we might be able to figure out what's causing the problem.

> I also have 2430548 memory left

Ahh, this is flash memory, so realistically it's not going to have an affect for you. What really matters is RAM. There is a RAM widget which might give you some idea of what the current RAM level is?

Shashank-Prasanna, 2020-09-24:
Oh,  never mind! It seems to be fixed when I moved to 2v07 **_AND_** made my own clock app instead of using Large Clock... I Don't know about Large Clock, but 2v07 sure seemed to help! Thanks! Also, I will check my RAM.

Shashank-Prasanna, 2020-09-24:
No, it is still there. I have 55 to 61% RAM, as the widget keeps giving me different numbers. I can't reproduce it, it just does it without warning! Here's the app page:
[image]
[image]

gfwilliams, 2020-09-24:
Did you change any settings on Bangle.js after you did the 'Install Default Apps'? 

And am I right that you're not using Gadgetbridge?

The fact that you don't have the latest timer app makes me think you never actually did 'Install Default Apps'? Could you try that, re-install the apps, and see if it stops the reboots?

Also, are you actually using the 'Battery Service' app? That's about the only one I can see might be problematic

Shashank-Prasanna, 2020-09-24:
Hm.. OK let me install default apps and get rid of BLE GATT

Shashank-Prasanna, 2020-09-24:
Seems to be fixed with reset, possibly an app/apps that are causing it, I will try to figure it out. Thanks!

gfwilliams, 2020-09-24:
Just a quick update - I literally just solved an issue that got picked up today:

If the Bluetooth stack needs restarting (eg. bluetooth services need adding) and it restarts at just the wrong time (literally - it has to be when a certain bit of the RTC is set) then the time can get put back 5 minutes which can cause a watchdog reset on the watch (if it reboots again it'd be really handy to know what appears at the top of the screen just after the screen goes blank).

Anyway, it was so soon after the 2v07 release that I just re-released with the fix in, so you could always try re-downloading and updating 2v07 and see if that helps.

Realistically you should only hit it if you'd enabled BLE HID in settings though.

Shashank-Prasanna, 2020-09-24:
Yeah it was enabled. Also, when it rebooted it said "Watchdog 2v05", and then the time got set back to 1970. Sad, because it is 2020. Also, I did have BLE on a lot of the time, so maybe it was that issue! I will go try it right now. Thanks!

Shashank-Prasanna, 2020-09-24:
Cool, it works! However, I won't close this issue just yet as I haven't tested it in the long run. I will close it sometime on Sunday if there are no issues. Thank you so much!

gfwilliams, 2020-09-25:
That's great!

Just a note though - you didn't actually have any HID apps installed, so realistically enabling HID in Settings wouldn't have been much use for you.

closed 2020-09-29 by Shashank-Prasanna