espruino/BangleApps #566: GadgetBridge sends mixed up info.

enaon, 2020-09-26:
I understand that this has little to do with bangle.js and is Gadgetbridge's issue, I just want to report that Gadgetbridge version 0.47 sends mixed up call event info.
Incoming call event sends cmd accept
Outgoing call event sends cmd Incoming
Start call event sends cmd reject
End call event sends cmd start

also the id that is send on notify- when a notification is read on the phone, does not match the id of the initial event, so there is no way to know which notifiction was read on the phone.

gfwilliams, 2020-09-28:
Thanks! I guess they changed the enumeration numbers for call states and then Bangle.js didn't get updated :(

https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L328

Very odd about the notification removal though - that definitely used to work. Also the weather must be a Gadgetbridge thing since we seem to be copying across the correct info: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L500

gfwilliams, 2020-09-28:
Ok, just fixed the call reporting issue anyway

enaon, 2020-09-28:
Nice :)

On the notification removal, I am not sure I understand what the id is supposed to do. I assumed that one can dismissmis/get informed about dismissal on all or speciefic notification. All works, specific one does not, as the id that is send on notify- , does not match any previouly send with notify. Maybe I got it wrong and the id is not event assosiated.

gfwilliams, 2020-09-28:
Yes, the notification ID should be the same for each notification, so you can see if a specific notification was dismissed.

enaon, 2020-09-29:
ok, I will check again to be sure.

enaon, 2020-10-03:
I did some more testing, GadgetbriDge 0.47.1, android 7.1.1. 

The nofication ID is working one way, the watch can dismiss a notification. But when a notification is dismissed on the phone, the id that is transmited on the dismissal is not the same as he one that was send on the arrival, it is on a different counter. 

So it is a problem only if one wants to keep syncronization, no big deal, I was just playing and noticed it. 
```
//notification 1 arrives
line:  GB({"t":"notify","id":1601425625,"src":"Hangouts","title":"xxx","body":"test"})
gb cmd in
//notification dismissed from watch
>gbSend({t:"notify", id:1601425625, n:"DISMISS"})
=undefined
//reply from phone
line:  GB({"t":"notify-","id":-287904279})
gb cmd in
//notification 2 arrives- dismissed on the phone
line:  GB({"t":"notify","id":1601425626,"src":"Hangouts","title":"xxx","body":"test2"})
gb cmd in
//send from phone on dismiss action, the ID is not the one that just arrived.
line:  GB({"t":"notify-","id":-287881850})
gb cmd in
```
Also, the calls are broken further than just mixed up info. I could not ignore or answer an incomming call, it seems that only reject works. A lot of things are not working and I am not sure if it is in my side.

gfwilliams, 2020-10-05:
Do you think 47.1 would have the changes I put in in it? Otherwise you'd expect the `cmd` to be wrong still.

Very annoying about the ID changing though - those numbers come direct from Gadgetbridge, and I'm pretty sure they worked in the past. I did spot a lot of refactoring having gone on though (which caused the `cmd` regression) so maybe something else got broken

enaon, 2020-10-05:
your changes are on the 47.2. not yet realeased but they are on the change log, so I will wait a bit to test them. 

I was hopping muting or aswering an incomming call whould work, even if I had to send something else. Will test again when 47.2 is released :)

enaon, 2020-10-16:
The 0.47.2 release is up so I tested the changes, call info is now fixed, thank you. :)

I followed the info on this page on sending from watch to phone:
https://www.espruino.com/Gadgetbridge
and out of those options:
```
t:"call", n:"ACCEPT/END/INCOMING/OUTGOING/REJECT/START/IGNORE"
```
it seems that I am only able to reject or end an incoming call. Answer or Ignore, seems to not work. Also I don't understand what incoming or outgoing is for when the message if from bangle to phone, maybe I am doing something wrong. Do you know if there is a bangle. js app that can uses gb call features so that I can see If I am missing something?

gfwilliams, 2020-10-16:
Ahh, it could be the docs are wrong there - it allows you to specify any enum that's defined in https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L187 but probably some of those aren't handled in Gadgetbridge

It seems like START should work though:

https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/receivers/GBCallControlReceiver.java#L63

enaon, 2020-10-16:
accept works on a xiaomi m11, thanks for the link.  :)

enaon, 2020-10-18:
accept/ignore works after all it seems on laterst android versions, it may be something on my phone that is wrong. Also the temperature send by weather is correct, it is in kelvin, my mistake. I havent tested the notifications id on a new android version phone, maybe they are working too, if I see that the don't I will make a comment later. 

So the gadgetbridge support is very nice, thanks. :)

gfwilliams, 2020-10-19:
Great, thanks for letting me know! Closing this then...

closed 2020-10-19 by gfwilliams