espruino/BangleApps #567: verticalface - Vertical watch face and GB notifications

dapgo, 2020-09-26:
App: Vertical watch face v0.05
FW: 2v07 (also in 2v05)
When I am on my prefer watchface/clock, GB notifications are not received on the bangle.js
However, using round clock or analog clock i receive them.
Is this watchface incompatible?

Another way to test is with the Gadgetbridge "Find Lost device"

detached, 2020-09-27:
I think the problem is that verticalface doesn't call `Bangle.loadWidgets();`.

gfwilliams, 2020-09-28:
Yes - unfortunately there's not enough space left in the vertical clock watch face for the widgets. It seems like a conscious decision of the clock designer.

I guess it could be moved around a little to make room though?

gfwilliams, 2020-09-28:
Just added widgets to the face - it'll be in the dev version of the app loader now

commit 9c5b4142 (2020-09-28): verticalface:  Tweak sizing to allow widgets at top, and add widgets (fix #567)
closed 2020-09-28 by gfwilliams