espruino/BangleApps #569: Vibrate clock display issue

berkenbu, 2020-09-29:
There is a screen update issue the new "Vibrate Clock" app when the hour switches from 2 digits to one (i.e. 23:59 -> 0:00 in 24 hour mode or 12:59pm -> 1:00pm). The new display string is constructed with a leading white space in these cases and "drawString" is called with with the "solid" option set to true; but the font used, 7x11Numeric7Seg, does not contain a white space character, thus no character is drawn and no background cleared for the first digit.
This leads, for example, to a transition 23:59 -> 20:00 in 24 hour mode.
This could either be solved by adding a whitespace to the font or by explicitly clearing the first digit via g.clearRect for times with hours < 10.
I can try and implement either, but would appreciate feedback which way is preferable.

gfwilliams, 2020-09-29:
Thanks! Actually adding a whitespace to 7x11Numeric7Seg seems like a great idea. It wouldn't add much extra memory usage and would solve this issue in other cases. I guess ideally Espruino itself should handle the space issue correctly when rendering, but that doesn't help the Bangles already out there.

There's the code used to create the font at the top of the file: http://www.espruino.com/modules/Font7x11Numeric7Seg.js  - so hopefully it wouldn't be too hard to modify.

berkenbu, 2020-09-29:
Modifying the font module was pretty straight forward, but I cannot for the live of me figure out in which repository it lives - trying to create a pull request... I have looked in the espruino and espruino-apps repo...

gfwilliams, 2020-09-29:
:) all the Espruino JS modules live in https://github.com/espruino/EspruinoDocs/blob/master/modules/Font7x11Numeric7Seg.js

berkenbu, 2020-09-29:
Thanks, added a pull request.

gfwilliams, 2020-09-30:
Thanks! Just merged. I'll update the espruino website in the next day or so and then the new modules should get pulled in and it'll be fixed

closed 2021-02-02 by gfwilliams