espruino/BangleApps #574: GPS recorder -recording during a long time

dapgo, 2020-10-08:
As GPS recorder allows to track in background, it is easy to forget to stop/turn off.

It is the second time that I keep **GPS recorder recording for a ¿too? long time, and after that I found that bangle.js is unresponsive.** 

The screen has a bit of backlight but it was not able to draw anything
Connecting to webIDE, Storage is empty
Connecting to bangle.js.apps, there is not installed apps.

I suppose that the tracking file went over some limits and maybe storage got corrupted.

It must be quite easy to reproduce, it happened on different FW versions.

Shashank-Prasanna, 2020-10-10:
A possible fix could be adding a corresponding widget. It would be helpful, as you could see directly if you're recording from your clock face. And @dapgo, you're not alone! I've done that many times, but luckily I caught it before the memory overflowed :).

gfwilliams, 2020-10-12:
> A possible fix could be adding a corresponding widget

The GPS recorder should add a widget though? A GPS satellite symbol when it is on?

Interesting about the memory overflow though - I'll see if I can reproduce it. It definitely shouldn't be able to mess up storage like that!

gfwilliams, 2020-10-12:
Ok, reproduced(ish).

When you realised the Bangle was unresponsive, what did you do?

What I think happened was Bangle.js realised storage was full but that some files had been deleted in the past, and then 'compacted' to free up some space to continue writing - which can take 30s or so.

However if you rebooted the watch (using BTN1+BTN2) during that time, storage would have been left corrupt, and when the watch came back up and checked it'd realise and erase everything in order not to brick itself.

Having said that even if you didn't reboot it yourself there's an issue with storage compaction right now (which I'm looking at fixing) which would mean the current app would likely hang until BTN3 was held

gfwilliams, 2020-10-12:
This issue was the one for compaction - just fixed now: https://github.com/espruino/Espruino/issues/1881

So now I'm having difficulty getting anything to break - apart from the big delay from Storage compaction - but I don't see we can do much about that.

dapgo, 2020-10-14:
> 
> 
> > A possible fix could be adding a corresponding widget
> 
> The GPS recorder should add a widget though? A GPS satellite symbol when it is on?
> 
> Interesting about the memory overflow though - I'll see if I can reproduce it. It definitely shouldn't be able to mess up storage like that!

Sorry, I was wrong, and yes the app has a widget that shows a satellite icon.
it is just that I was not associating this icon with GPS recorder but with the signal and usage of GPS.
I suppose that i was expecting an icon with red color to express recording. :)

gfwilliams, 2020-10-15:
Yeah, I think there's a little red aerial animation?

I did actually have a big red circle in the past, but someone complained that they didn't know what it was, so I changed it to a satellite icon. Can't make everyone happy!

I guess we could have a satellite icon with a flashing red circle instead of the aerial sign if you want to make a PR? :)

The only other thing we could do now is to trap the exception `StorageFile.write` causes when the Storage is full, and stop recording/put up an icon saying storage is full

hughbarney, 2021-01-01:
Nice App - looking forward to trying when I get my watch post Jan 4th 2021.  Nice to see screenshots in the github repository so I can see what the App looks like.

commit 15be69ba (2021-02-02): gpsrec 0.17: Disable recording if storage is full (fix #574)
closed 2021-02-02 by gfwilliams