espruino/BangleApps #577: simpletimer fails with "Buzz in progress" error

Franzo-, 2020-10-14:
When the timer is in "stopped" state (i.e. "Time UP!" is written on the screen) if BTN1 is pressed while the watch is vibrating, the next time the timer goes off the buzzer does not vibrate and the following output is written on console:
```
Uncaught Error: Buzz in progress
 at line 11 col 29
  return Bangle.buzz(1000, 1)
                            ^
in function "buzzAndBeep" called from line 21 col 15
  buzzAndBeep();
              ^
in function "outOfTime" called from line 49 col 15
    outOfTime();
              ^
in function called from system
```

The issue is easily reproducible even using the emulator.

From what I can understand, the condition to reproduce this bug is very similar to #521 and analyzing the Espruino code I saw that the function `jswrap_banglejs_buzz_callback()` is expected to be called after a timeout set by `jswrap_banglejs_buzz()`. If all timeouts are cleared by a call to `clearInterval()` without arguments from JS code, even this timeout is removed and the callback is never executed. This causes the exception on the next `Bangle.buzz()` call.

There is an easy and viable workaround for the simpletimer app, which is storing all the references of timeouts and intevals created with `setTimeout()` or `setInterval()` and clearing them explicitly in the `clearIntervals()` function. This can work for this app, but this is a problem which can unexpectedly arise in every JS user code. If possible, I think it should be better to fix it in the Espruino library code.

gfwilliams, 2020-10-15:
Thanks - I think either way a fix for the timer app is a good idea anyway - it's just good practice and would fix it for existing watches. It may even be we need to change a tutorial as well.

But in terms of the buzz/etc I guess we'd have to be able to mark specific timeouts as 'system' timeouts so that a simple `clearInterval` wouldn't get rid of them. It's definitely possible.

closed 2020-10-15 by gfwilliams

Franzo-, 2020-10-15:
Thank you for your quick reply!
Unfortunally you missed to clear the timeout set in `buzzAndBeep()`, so now the periodic buzz can never be stopped...

I propose to use this code in the app:
```
let buzzInterval;

function buzzAndBeep() {
  buzzInterval = -1;
  return Bangle.buzz(1000, 1)
    .then(() => Bangle.beep(200, 3000))
    .then(() => { if (buzzInterval) buzzInterval = setTimeout(buzzAndBeep, 5000); });
}
```

And in clearIntervals():
```
function clearIntervals() {
  if (alarmInterval) clearInterval(alarmInterval);
  if (counterInterval) clearInterval(counterInterval);
  if (buzzInterval && buzzInterval > 0) {
    clearInterval(buzzInterval);
  }
  alarmInterval = undefined;
  counterInterval = undefined;
  buzzInterval = undefined;
}
```

In this way `buzzInterval` is cleared only when it is set and `buzzAndBeep()` should not set the timeout again if `clearIntervals()` has been called while the buzzer or the beeper are on.

gfwilliams, 2020-10-16:
Thanks! Just added that and it seems ok now

commit 315e7c6b (2020-10-15): simpletimer: Fix buzz error, remove '+' when timer running and add 'back' text (fix #577)