espruino/BangleApps #579: fileman/filemanager

dapgo, 2020-10-15:
v.001
fw 2v8

Opening the app it seems to crash when there is a ¿"too big"? file in storage
banglerun_201001_122502 (StorageFile)

I don't know exactly how big was the file, because  download from IDE, Storage was also failing.
But before deleting it I read the file from IDE line by line and it printed 120lines (char width <73)

I am not sure it all content was printed as i receive also a 
"Prompt not detected - upload failed. Trying to recover..."

gfwilliams, 2020-12-08:
Thanks - just fixed this - ish.

It's related to the way StorageFile chunks appear in the normal file list. This will improve matters, but it's still not perfect and could well fail if there are several StorageFiles.

Hopefully the situation will get changed in a Bangle.js firmware soon.

closed 2020-12-08 by gfwilliams

gfwilliams, 2020-12-09:
Ok, new Bangle.js firmware will fix this by only listing the *first* StorageFile chunk. Also it uses bigger StorageFile chunks so should be better there too. So firmwares built after this date, or 2v09+ will have it

commit 6a63a6ea (2020-12-08): fileman 0.02: Improve handling of large amounts of files (fix #579)