espruino/BangleApps #582: Graphics API changes

gfwilliams, 2020-10-20:
There are a few things I'm not happy with right now:

 [ ] We reserve 24px of space at the top of the screen for widgets, however it'd be nice to allow a thin widget bar for those that care
 [ ] If there are no widgets at the bottom it'd be nice to ensure apps could easily tell and use the extra space
 [ ] It'd be nice to allow a global color theme, covered in https://github.com/espruino/BangleApps/issues/499
 [ ] there's no way to request an app redraws everything (eg maybe we displayed a notification and lost what was on screen)
 [ ] I think Bangle.setLCDMode for all but double buffering should be removed, and standard `Graphics.drawImage` with an offscreen buffer should be used instead as it's way more flexible.

Any thoughts about these?

* I guess we could have `g.clientArea => {x,y,w,h}` which we updated when drawing widgets *and* the app, so then apps/widgets would have a clear idea what area they were expecting to redraw?
* `g.theme.bg/fg/bg2/fg2` for the theme?
* Maybe a `Bangle.on('redraw'` event apps could hook onto? Notifications could also check to see if the event was implemented before trashing the screen

rigrig, 2021-03-17:
Wouldn't having a thin widget bar also mean that all widgets need to check their height, and then fit themselves into whatever space there is?

`g.theme.` sounds nice, what would you use `bg2` for though? Maybe just have one `bg` and `fg1`,`fg2`,`fg3`?
I'm not too sure about changing `g.reset()`, it wouldn't surprise me if there are apps doing `g.reset().setColor(-1)`, and they only show up as broken for people with black-on-white themes.

`Bangle.on('redraw')` seems nice, although I'm not too sure how useful it will actually be: notifications can't really trash the screen anyway, because the clock might draw over it every second. Unless there is some way to stop other apps from drawing?

gfwilliams, 2021-03-18:
> Wouldn't having a thin widget bar also mean that all widgets need to check their height

Yes - although in many cases the change would be pretty simple. It's actually more for targeting different watches/devices that have different screen resolutions - 24px is a bit large on some of them.

> what would you use bg2 for though?

Imagine a menu where you have a highlighted entry. You want a different foreground but also a different background.

> it wouldn't surprise me if there are apps doing g.reset().setColor(-1), and they only show up as broken for people with black-on-white themes.

True, but not doing that means you definitely have to change *all* apps/widgets. The plus side I guess is it could be implemented in a backwards-compatible way.

> Bangle.on('redraw')

> Unless there is some way to stop other apps from drawing?

Yes - you can do clipRect or 'setLCDMode("null")`. The current notifications scroll the screen and use a different area of the screen buffer.

The other benefit of this is you could actually get a proper screenshot capability - by having a small (240 x 24 x 16?) buffer and then repeatedly calling `Bangle.emit('redraw')` you could get a full image of the screen back to the PC (albeit slowly!)

rigrig, 2021-03-27:
I guess it would be nice if widgets could scale. Its just that I expect getting all widgets to play nice with a variable drawing area to be a lot of effort.

> You want a different foreground but also a different background.

Right, that makes sense.
I was thinking more along the lines of having different colors, e.g. you would draw most things in `fg1`, but maybe add some elements in `fg2` so it isn't all monochrome. But I don't think many apps use two colors now anyway, so that might be a bit overkill.

>> it wouldn't surprise me if there are apps doing g.reset().setColor(-1)

On reflection, this probably won't be much of a problem, as we will searching to change every `setColor(-1)` anyway. And needing to call `.setColor(g.fg).setBgColor(g.bg)` after every reset actually doesn't sound all that great.

gfwilliams, 2021-12-02:
We now have `Bangle.appRect` - it's just getting apps to use this

gfwilliams, 2022-11-30:
Thanks! Yes, I think that makes sense.

The one remaining one I see is the `redraw` option - and it strikes me that we could add that to `Bangle.setUI` pretty easily?

Also it's not such a big deal as we have overlays now

gfwilliams, 2024-05-02:
Closing this - we now have `redraw` and while setLCDMode is still there it's not really used

closed 2024-05-02 by gfwilliams