espruino/BangleApps #585: Reconnection with gadgetbridge and web app loader

Grrrolive, 2020-10-27:
Hello, 
Hope this hasn't already asked, as I didn't found any relevant issue before. 

Now using gadgetbridge for notifications, very useful and practical on Android 10. 

When I want to do something on the web app loader, I see the button "disconnect", but the app list refreshes and never comes. 
In addition, if I push on "disconnect" and then try to "connect", the pop-up appears, but never shows my bangle watch. 

I have found this solution : 
Go in gadgetbridge, disconnect manually. 
Then, go in the android Bluetooth menu and disconnect manually the bangle watch. 

At this time, the connecting process is ok, it founds the watch in few seconds, and all seems to work. 

Is it possible to make it easier ?

gfwilliams, 2020-10-28:
Hi,

Glad you found a way around it, but yes, you're right - the connection to Gadgetbridge gets in the way of the Web Bluetooth connection.

I'm not really sure there's a way around it, apart from my extending Gadgetbridge to allow you to access Web Bluetooth through the app?

Grrrolive, 2020-10-29:
I didn(t know there was an extension for gadgetbridge. 
Where could I find it ? 
Or how to install it ?

gfwilliams, 2020-10-29:
> I didn't know there was an extension for gadgetbridge

Sorry. There isn't! I was saying that's probably the best way around it without requiring a disconnect - however I have asked the Google Chrome Web Bluetooth team if they have any ideas - it might be possible for Chrome on Android to allow connections even when Gadgetbridge is already connected.

Grrrolive, 2020-10-29:
Ok, sorry, my understanding was not correct as I am a french guy, my english is not perfect. 
Indeed, this option would be really interesting. 
And what about mozilla firefox ?

gfwilliams, 2020-10-30:
> And what about mozilla firefox ?

I'm afraid Firefox refuses to support Web Bluetooth at the moment, so that's not going to work :(

closed 2020-10-30 by Grrrolive