espruino/BangleApps #588: Heart Rate recorder view records

midnight4577, 2020-11-11:
v.002
fw:2v08.82
 
 
I want to use this app to record my heart rate while doing stuff like running and it does that perfectly, but I can't veiw the recorded heart rates without opening my pc, connecting to my watch via blutooth, and then downloading the data from the app loader.

I was thinking that the app could be made to graph the data, kind of like the gps recorder.
can anyone help me out with this?

Ps. I'm not that good at Java script.

gfwilliams, 2020-11-12:
Hi, that sounds like a good idea.

Looking at the code, it actually looks like it's based on the GPS recorder, so it should almost be copy & paste to add graphing...

Heart Rate recorder: https://github.com/espruino/BangleApps/blob/master/apps/heart/app.js#L62
GPS recorder: https://github.com/espruino/BangleApps/blob/master/apps/gpsrec/app.js#L121

So I'd suggest you take the bit where heartCount/heartTime are calculated and put it into a function called `getInfo`, then copy/modify the `plotGraph` function from the GPS recorder to plot the heart rate.

the GPS plotGraph has some code for `altitude` that's pretty much what you want. The data written is `time,heartrate,confidence`, so instead of:

```
    title = "Altitude (m)";
    while(l!==undefined) {
      ++nl;c=l.split(",");
      i = Math.round(200*(c[0]/1000 - strt)/dur);
      infn[i]+=+c[3];
      infc[i]++;
      l = f.readLine(f);
    }
```

You want something like:

```
    title = "Heart rate";
    while(l!==undefined) {
      ++nl;c=l.split(",");
      i = Math.round(200*(c[0]/1000 - strt)/dur);
      infn[i]+=+c[1];
      infc[i]++;
      l = f.readLine(f);
    }
```

The code there's a bit cryptic, but it's just because it was designed to run reasonably fast because it's having to chew through a lot of data.

Hope that helps!

midnight4577, 2020-11-12:
Thank you so much, I will try and get it working with my limited Java script knowledge.

Andrwe, 2021-04-10:
@midnight4577 I implemented code to graph the last 164 records (matches around 20 minutes) of a file. This was merged and is available in v0.0.3.

Sometimes it might result in OOM and due to kill of the app which is why I'm working on improving the memory usage right now.
But for small files (up to 4 hours of data) it works.

midnight4577, 2021-04-10:
Ok, I've tested it and it appears to be working fine, Thanks.

By the way the graph looks really good :)

closed 2021-04-10 by midnight4577