espruino/BangleApps #609: [New Feature] For worldclock

hoekma, 2020-12-11:
@computermacgyver  -- I made an update to the worldclock source code to make the secondary time display larger if there is only one timezone active in the array.  I don't have access to push a branch / PR but can share it with you if you like.  It would make my life a lot easier given my 52 year old eyeballs!  :)

computermacgyver, 2020-12-11:
Great idea @hoekma . You can fork this repository to your own GitHib account, make the change, and then open a PR. The steps are the first and last parts of [this page on forking and submitting a PR](https://www.espruino.com/Bangle.js+App+Loader). I'm also happy to get your modifications another way and help polish anything

hoekma, 2020-12-11:
@computermacgyver Done!  Please see https://github.com/espruino/BangleApps/pull/610

computermacgyver, 2020-12-13:
Thank you. Please feel free to close this issue @gfwilliams

gfwilliams, 2020-12-14:
Thanks! Just merged #610

closed 2020-12-14 by gfwilliams