espruino/BangleApps #611: intl: use Ecma402 API where possible

srl295, 2020-12-14:
ideally, use standard https://github.com/tc39/ecma402 

For example. `new Date().toLocaleDateString()` instead of `require('locale').date(new Date())`

could also have a shim `Intl` object. 

See also #202  and #136 and #304

gfwilliams, 2020-12-15:
> For example. new Date().toLocaleDateString()

This doesn't exist in Espruino (which is why we added the Locale library), so there would have to be some kind of polyfill for this.

Do you have an example of what the equivalents of http://www.espruino.com/Bangle.js+Locale would look like?

srl295, 2020-12-15:
> > For example. new Date().toLocaleDateString()
> 
> This doesn't exist in Espruino (which is why we added the Locale library), so there would have to be some kind of polyfill for this.

Right.

> Do you have an example of what the equivalents of http://www.espruino.com/Bangle.js+Locale would look like?

That was an example.  `new Date().toLocaleDateString()` will do roughly what `require('locale').date(new Date())` does.

gfwilliams, 2020-12-15:
That's one example, which doesn't use `Intl`.

What about distance, speed, currency and temperature?

srl295, 2020-12-15:
> That's one example, which doesn't use `Intl`.

it's the convenience version, Could also be something like `(new Intl.DateTimeFormat()).format(new Date())`

> What about distance, speed, currency and temperature?

from https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Intl/NumberFormat#Using_style_and_unit - 

```js
console.log(new Intl.NumberFormat("pt-PT",  {
    style: 'unit',
    unit: "mile-per-hour"
}).format(50));
// → 50 mi/h
```

gfwilliams, 2020-12-15:
Ok, that's kind of what I thought... So how would the app developer know what to set the locale and unit to?

The idea of `locale` was very much that all that was handled for the developer. They just say "I have a speed here, display it in a way that the user understands".

srl295, 2020-12-15:
> So how would the app developer know what to set the locale and unit to?

there are ways to do that. - see https://unicode-org.github.io/cldr-staging/charts/38/supplemental/unit_preferences.html#meter-per-second for example 

What's the best way to give some comments on the i18n aspects of the `locale` class? An issue?

gfwilliams, 2020-12-16:
Thanks. I'm still curious to see an actual code example of the equivalent of something like `require('locale').speed(150)` using the `Intl` class though.

> What's the best way to give some comments on the i18n aspects of the locale class? An issue?

Yes, that's probably best.

However as I mentioned, I've put a decent amount of work into the `locale` class (with input from those that requested it) and it's hardly being used (even by those that asked for it) - which means that personally putting more work in is not high up my priority list.

If you're willing to add internationalisation to a bunch of existing apps with your suggested changes in then that'd be great though and I'd be extremely happy to take PRs.

closed 2021-04-14 by gfwilliams