espruino/BangleApps #613: Digital Assistant, not EDITH, counter is now chnaging with battary level over buttons

PHDPeter, 2020-12-21:
Hi
Not sure who is working on the Digital Assistant, not EDITH app but I think that the "count" variable has been overused and is now being updated with the battery level, not just with the buttons.
I have a feeling it is line 70 that is the issue? "count = l" where l is the battery level.
Maybe it is worth having this as an option in the setting as I am sure some people would love to have the big count show the battary.
Sorry if this all meant to be like that, I am just starting to code with the bangle.js so please ignore if it is meant to be this way.

gfwilliams, 2020-12-22:
Thanks for the report!

@OmegaRogue any ideas about this?

OmegaRogue, 2020-12-22:
Both the original button counter and the current battery display are only temporary stages, in the end the counter should be changed via bluetooth or maybe also buttons, but the button counter will be back soon

PHDPeter, 2020-12-31:
@OmegaRogue  that sounds good :-) love the watch face btw

PR #632 (2021-02-17): :rewind: Release DANE 0.16
### Release DANE Watchface version 0.16

- [x] Testing

#### Changes:
- Revert Counter to being button controlled, as the previous batters display was for debug purposes while implementing the battery bar

Closes #613
closed 2021-02-17 by gfwilliams