espruino/BangleApps #615: Retain time while manipulating user interface

OhMeadhbh, 2020-12-26:
When pressing buttons, the watch randomly resets the time to 0 ( Midnight at 1 January, 1970 ).

Is there a way to have button presses not reset the time?

OhMeadhbh, 2020-12-26:
Also, GPST is several seconds off UTC. Is there a way to use UTC time instead of GPST?

gfwilliams, 2020-12-31:
Are you sure your watch and app firmware is up to date? What apps do you have installed? And was it one specific app that made it lose the time?

The watch should definitely not reset itself to 1970. About the only way you can do that unless something is broken is by holding BTN1 and BTN2 for 10 seconds which does a hard reset. And even then, if you'd even had a GPS lock when it reboots it should be able to pull the right time off GPS

OhMeadhbh, 2021-01-01:
I'll be on the lookout for new firmware. The watch came with 2v08.

After resetting several times and not wearing it, it seems to only reset once or twice per day.

Also, GPST is different from UTC. UTC includes leap seconds that are not used in GPST, which is why there's a 18 second difference between the two. What I was implying with my second comment above is it would be nice if the watch could be set to UTC and not GPST.

OhMeadhbh, 2021-01-01:
Ah. Even though it looks like the 2v08 was installed on the watch, I went ahead and re-installed 2v08 from the ZIP file downloaded from https://www.espruino.com/Download#banglejs . I appreciate that the process is well documented at http://www.espruino.com/Bangle.js#firmware-updates - the update itself went off without a hitch.

It absolutely reset the time to the beginning of the Unix epoch after installing the firmware again.

It did, however, keep the default clock setting and dutifully booted into the Vertical Face clock which I set as the default before updating the firmware.

But the vertical face clock now shows eternal 00:00 Thu 1 01 - it's not advancing at all. The Morphing Clock is updating, however, but it's showing several minutes past midnight on Jan 1 1970. I noticed this comment ( http://www.espruino.com/Troubleshooting+Bangle.js#i-can-t-get-a-gps-fix-gps-doesn-t-seem-to-be-working ) suggesting I try at least 10 minutes outside to get a proper fix on GPS.

I'm now wondering if maybe the problem originally with the time being off was it was getting a partial fix (I was sitting inside when this happened after failing to get a solid lock after 10 minutes outside.) and the GPS hardware / firmware was sending a series of updates as it acquired and lost proper lock.

Let me go outside for 15 minutes and see if I can get a solid GPS lock. I'm under partial canopy, but this reference suggest 15 min should be good. ( https://www.tandfonline.com/doi/pdf/10.5721/EuJRS20134636 )

Measured PDoP at my latitude is never perfect, but generally always less than 3. I've never tried to measure or calculate TODP.

After 15 minutes outside it failed to acquire GPS time.

Thinking I should reset the device and THEN set it outside for 15 minutes, I reset it. It most definitely reset the time to midnight new-years 1970. After another 15 minutes, it still failed to acquire GPS time.

If this is in any way useful for debugging, when reset, it displays "Searching for GPS time" for just less than a second before displaying the UI. Is it trying to find GPS, failing and then giving up? Or is it still trying to find GPS in the background?

OhMeadhbh, 2021-01-01:
Closing Issue. Device is clearly defective and the software is a little too alpha for what I wanted to do.

closed 2021-01-01 by OhMeadhbh

gfwilliams, 2021-01-01:
It sounds to me from one of your other posts like you're holding BTN1+BTN2 (top right & middle) to get back to the main clock face? That's like the 'last ditch' reset, which resets everything including the time. If you hold BTN3 (bottom right) for just a few seconds it'll move back to the clock app properly - without losing the time - that's what should normally be used.

> If this is in any way useful for debugging, when reset, it displays "Searching for GPS time" for just less than a second before displaying the UI. Is it trying to find GPS, failing and then giving up? Or is it still trying to find GPS in the background?

Yes, that's expected. It just turns GPS on and waits for it to boot and then asks it the time - it doesn't actually leave the GPS on for long enough for it to get a lock. There is a widget that allows it to do that though.

If you're not getting a GPS lock while using a GPS app, you'd probably find installing the AGPS app helps significantly - it sends a bunch of up to date info to the GPS receiver so it can get a lock much more quickly - much like your phone is able to do (since that has an internet connection).

I'm on holiday until 4th Jan, but after that I'll be able to provide you with a lot more help to get this working.