espruino/BangleApps #616: Update time in clock app

OhMeadhbh, 2020-12-26:
I noticed that if I let the screen blank, and then press a button to turn the screen back on, the time in the current clock app does not update. (I tried it with the default Morphing Clock and Vertical Face clock apps.) If I press a button to keep the screen active, the time does update. Exiting the clock app and re-entering a different app occasionally causes the newly launched clock app to display the correct time, but it also usually results in the time resetting to 0.

gfwilliams, 2020-12-31:
Are you sure all your clock firmware is up to date? This should totally not happen - if you wake the clock app then the time displayed should be the current time - although for some clock faces there might be a very small delay as it updates the time.

OhMeadhbh, 2021-01-01:
How do you prevent the watch from resetting to 1970?

Or if you can't do that, how do you coax it to get GPS lock? I sat in a field today for 30 minutes waiting for it to find GPS. Is there a setting somewhere? I have a bus-pirate around here somewhere, so I could open up the back and try to connect via JTAG / OpenOCD, but that seems like a bit of overkill.

OhMeadhbh, 2021-01-01:
Closing Issue. Device is clearly defective and the software is a little too alpha for what I wanted to do.

closed 2021-01-01 by OhMeadhbh