espruino/BangleApps #618: About app shouldn't hang UI

OhMeadhbh, 2020-12-26:
The current behaviour of the About app is to hang the UI, so you have to reset the device to get back to the main menu. It would be nice if there were a way to cleanly exit the About app. Every time I reset the device, I have to manually reset the time, which is annoying.

OhMeadhbh, 2020-12-26:
Several of the clocks seem to default to a long-press of button 2 to exit the app and return to the list of apps. About should probably do the same thing.

gfwilliams, 2020-12-31:
Hi - all you should have to do is long-press BTN3 which is the standard way to exit all apps and return to the main clock.

But yes, it's a good point - we should probably make any button press return to the clock from the About page.

>  Every time I reset the device, I have to manually reset the time, which is annoying.

How do you reset the device? If you do it with BTN3 the time should definitely not be lost.

OhMeadhbh, 2021-01-01:
The time is most definitely reset to 1970 on each reboot.

To reset the device I press button 1 & 2 for about 10 seconds, then release the buttons. From a reset, I *can* go into the DFu mode, but it doesn't seem to make a difference in resetting the clock.

OhMeadhbh, 2021-01-01:
Closing Issue. Device is clearly defective and the software is a little too alpha for what I wanted to do.

gfwilliams, 2021-01-01:
> To reset the device I press button 1 & 2 for about 10 seconds, then release the buttons

Ok, this seems like it's your problem. To get back to the clock, just hold BTN3 (bottom right) for around 3 seconds.

That'll go back to the clock without resetting the time, and is much quicker and easier.

BTN1 + BTN2 is pretty unwieldy and is there as a last-ditch way to completely reset Bangle.js if everything has broken. Realistically it should never actually end up being used (I can't remember the last time I've had to use it). 

Did some of the docs give you the impression BTN1+BTN2 was the thing to do? If so we should definitely change that - as you say, the experience of trying to use that to change apps is pretty poor, especially as it resets the time on each reset!

gfwilliams, 2021-01-01:
And reopening this since i think it's fair to have a button that returns from the About page

dapgo, 2021-01-02:
I would also add a touch event. 
In fact every button or touch area can be set to go back, as from being the about page a single page, the only expected behavior is going back.
But maybe a 2 "pages" app should be considered, as now characters are quite small.

commit 457ccb1d (2021-02-01): about 0.07: Pressing a button now exits immediately (fix #618)
closed 2021-02-01 by gfwilliams