espruino/BangleApps #628: route viewer

dapgo, 2021-01-10:
I think this app has a lot of potential with some improvements

- import .GPX function
    the .KML files are good but most android apps only manage and export .GPX files.

- zoom add a zoom feature

- make a normal app+widget or integrate routerviewer on GPSrecorder 
 Making possible to import different tracks/maps and have them available from the app menu

gfwilliams, 2021-01-11:
PRs very welcome ;)

Actually parsers for KML and GPX would be a good thing to add as a common library as I imagine they might be pretty useful.

Also you can now use the OpenStreetMap app as a library, so it's reasonably easy to draw a map as a background

dapgo, 2021-01-15:
I would like to do some improvements myself, but besides the lack of time I need also to install and spend a bit of time with github desktop.

I have couple of apps/corrections pending to be uploaded to the repository, once i tried from the web and i created a mess.

gfwilliams, 2021-01-15:
Ok, thanks.  If you do have corrections then if you've got them online somewhere - even if it's only as the JS file that you upload - you could paste them as an issue here and I'll manually pull them in

bernhardreiter, 2021-06-15:
> Actually parsers for KML and GPX would be a good thing to add as a common library as I imagine they might be pretty useful.

Did a quick search and found https://yarnpkg.com/package/@tmcw/togeojson, which advertises: "convert KML, TCX, and GPX to GeoJSON"

gfwilliams, 2022-02-11:
Did a minor update to make it run on Bangle.js 2

closed 2024-05-02 by gfwilliams