espruino/BangleApps #631: App Loader Broken (for me at least)

Shashank-Prasanna, 2021-01-12:
Hello everyone, 
I was downloading apps today and I got the error `Device connection failed, Invalid JSON` while getting the app list. It worked earlier in the day. I deleted all of the apps, and it still gives me that. Now the Bangle does basically nothing because it doesn't have a bootloader. Welp. So what happened? Why? How can I fix it? Also, there was an error where even if I downloaded AGPS multiple times, it didn't show up or work, maybe that caused it?? Thanks in advance for any help!

gfwilliams, 2021-01-13:
Hi - I believe what has happened isn't the app loader, but somehow the data describing the apps on your watch has got corrupted.

Any chance you could bring up the Chrome Developer Console and paste up the log messages you get when trying to download an app? It might help us find out the culprit.

Can you do `About -> Install Default Apps`? That will end up deleting everything from your Bangle but should fix it.

If not, you'll definitely be able to connect with https://www.espruino.com/ide/ then paste the command `require("Storage").eraseAll()` on the left-hand side and that'll fix it - but obviously that will remove all apps as well.

> I downloaded AGPS multiple times, it didn't show up or work

It's not actually meant to show up, since it just pushes data to the GPS. I'll update the description to make that clearer.

Shashank-Prasanna, 2021-01-13:
[image]
So...didn't upload, just kept saying `Removing Apps` over and over again. I'll post another comment with the IDE command later. Also, IMPORTANT: I uploaded the bootloader yesterday. I rebooted, and then, when it was on the Bangle.js screen, it got the app list and worked. That's the only app on it.

Shashank-Prasanna, 2021-01-13:
> If not, you'll definitely be able to connect with https://www.espruino.com/ide/ then paste the command `require("Storage").eraseAll()` on the left-hand side and that'll fix it - but obviously that will remove all apps as well.

Nope, didn't work. Both the desktop and website versions didn't work :(. On the first try, it said unsupported device. Second, try, it worked but said `No Response`. When I copied and pasted the command, it didn't show up in the console. The words `Sending` and `Sent` quickly flashed on the console, but the bootloader is still on my Bangle and it still doesn't work.

gfwilliams, 2021-01-14:
Ok, sounds like maybe the Bangle itself has hung? Either that or you could have turned off some of the Bluetooth connectivity settings like `Programmable`.

Can you try this please? https://www.espruino.com/Bangle.js#resetting-without-loading-any-code

Then connect with the Web IDE and see if you can paste the code?

Shashank-Prasanna, 2021-01-14:
It worked! YAY!!!! So resetting it and connecting seemed to work.

closed 2021-01-14 by Shashank-Prasanna

gfwilliams, 2021-01-15:
Great! If this happens to you again can you let me know? It'd be good to try and figure out what happened so I can make sure it doesn't happen again

Shashank-Prasanna, 2021-01-15:
@gfwilliams sure! Although you might want to start. by checking Digiclock. It seems to have some issues

Weiming-Hu, 2021-03-31:
Hi, I have a similar problem here. Please see my screenshot below.

[image]

When I click refresh under my apps, it usually works. But I don't know why it has this issue at the first place. Is there any way I can debug this?

Thank you

Shashank-Prasanna, 2021-04-04:
It seems as if the JSON is corrupted. Try deleting new apps that you suspect might be the problem, and if that doesn't work, reset your Bangle.

Weiming-Hu, 2021-04-04:
I'm actually developing my own app. So I'm not sure whether my app is causing this. Is there anyway I can debug? Or see why and how JSON is corrupted? Thank you.

gfwilliams, 2021-04-06:
Does it work if you boot Bangle.js so it's just showing the logo? It could be that the clock takes a while to render or does something that means the code fails with an 'out of memory' error or similar?

Hi! You can type `Puck.debug=3` in the web developer console, then you should be able to see what's being transferred.

You can also breakpoint the code in `comms.js` to see what comes back.