espruino/BangleApps #693: GPS Widget and HRM Widget stopped the clock loading correcly

PHDPeter, 2021-03-17:
Just a quick bug I found where my bangles clock would not load correctly with either of the apps loaded.
Some times it would load (but the bt 2 push would then not work for the launcher) but most of the time is just loaded a black screen where the clock should be. Widgets were still there but the launcher for apps would not load from the push of bt 2.
Have not looked at the code but thinking that it might just be messing with the running of the clocks as that would explain the bt 2 not working?

PHDPeter, 2021-03-17:
I was using on the 2v08 of the firmware, did not test with any other firmware,

gfwilliams, 2021-03-18:
Ahh - yeah, the widgets both say `Requires firmware v2.08.167 or later`

Obviously it'd be nice if they didn't totally break everything on older firmwares though! I've just pushed a fix and will get this into the official BangleApps soon

closed 2021-03-18 by gfwilliams

PHDPeter, 2021-03-18:
Aww that's ok then :-) thanks

commit 84372932 (2021-03-18): don't break the whole watch if widget is installed on 2v08 and older firmwares (fix #693)