espruino/BangleApps #700: a centralized widget configuration

dapgo, 2021-03-21:
I think that it would be great to have a function to administer every widget installed.

Functions can be: widget area (top, down), sort order, hide/show, enable/disable..

I suppose it can  be also a fork of "App Manager" but focused in widgets

On the other hand, on [README.md#graphic-areas](https://github.com/espruino/BangleApps/blob/4c78076603b36bcccff102ca2989c5ad82b8924b/README.md#graphic-areas) there is no reference to bottom widget area

gfwilliams, 2021-03-22:
I'll update the graphic areas bit - but yes, an app manager would be pretty neat.

I think there was a discussion on the forum about it, but it should be pretty easy to do something. The app could add a `widgetconf.boot.js` file that overwrite `Bangle.loadWidgets` with a file that loaded widgets but then overwrite the `area` element and also re-ordered them as required.

gfwilliams, 2022-02-03:
Fixed with https://banglejs.com/apps/#wid_edit

closed 2022-02-03 by gfwilliams