espruino/BangleApps #711: Mouse HID through bluetooth

midnight4577, 2021-03-28:
I was recently working on a small Mouse app that I'm hoping should turn the bangle's tilt and buttons into a simple mouse. But it appears that even though the module seems to work the watch needs support for the mouse in it's HID settings. I have no idea how to add this (although I did see that there appears to be some HID stuff in the boot code) as I suck at JS and do not have much understanding of the watch's inner workings.

Any help would be appreciated.

This is the Mouse module: [https://www.espruino.com/modules/ble_hid_mouse.js](https://www.espruino.com/modules/ble_hid_mouse.js)

Here is my simple code:
<pre><code>
//simple bluetooth mouse

var storage = require('Storage');
var mouse = require("ble_hid_mouse");

const settings = storage.readJSON('setting.json',1);

NRF.setServices(undefined, { hid : mouse.report });

function btnPressed() {
  g.reset();
  g.clearRect(50,50,100,120);
  g.drawString("failed", 50, 50);

  mouse.send(0,0,mouse.BUTTONS.LEFT); // X movement, Y movement, buttons pressed

  g.clearRect(50,50,100,120);
  g.drawString("Success", 50, 50);
}

if (settings.HID=="kb" || settings.HID=="kbmedia") {//need to change this as soon as i find a way to add a Mouse HID setting

// trigger btnPressed whenever the button is pressed
setWatch(btnPressed, BTN, {edge:"rising",repeat:true,debounce:50});

} else {
  E.showPrompt("Enable Mouse HID?",{title:"HID disabled"}).then(function(enable) {
    if (enable) {
      settings.HID = "kbmedia"; //if I find out how to add a setting for Mouse HID I need to change this
      require("Storage").write('setting.json', settings);
      setTimeout(load, 1000, "hidmouse.app.js");
    } else setTimeout(load, 1000);
  });
}
</code></pre>

gfwilliams, 2021-04-06:
Hi! The code to set the HID details is at: https://github.com/espruino/BangleApps/blob/master/apps/boot/boot0.js#L8

All you'd need is to add:

```
else if (s.HID=="mouse") Bangle.HID = E.toUint8Array(atob("....."));
```

And you get what's in "...." by doing `print(btoa(require("ble_hid_mouse").report))`.

Hope that helps! I'd rather not add the HID report code right now until you have something that works for you, but when you're sure it's working then I'll happily merge

midnight4577, 2021-04-06:
I have successfully added the HID option in the boot code and uploaded it via espruino but the mouse still does not work, The watch is disconnected from the espruino IDE and connected to my computer through BLE, HID mode is set to mouse, and there does not seem to be any errors, my computer even thinks that the watch is a mouse but I cant seem to get it to move or click. any idea of why it's not working? I think it might be something with what is actually sent through BLE and I might try just sending something through `NRF.sendHIDReport()`

gfwilliams, 2021-04-07:
Ok, that's interesting. I forget where but I saw someone else having issues with the HID mouse. What OS do you have?

So it might be that we need to come up with a different HID report style that'll work across more OSes :(

midnight4577, 2021-04-07:
I'm currently using Windows 10 but I'm intending to use the mouse with a raspberry pi.

I might look into building or modifying and existing HID mouse report but as I am currently in school I will not have much time to research what that involves and how to do it.

gfwilliams, 2022-02-03:
This is now fixed. We have a `"com"` type for HID in Bangle.js settings that provided both Keyboard and Mouse

closed 2022-02-03 by gfwilliams