espruino/BangleApps #716: gadget bridge - sendActivity improvment

valvin1, 2021-04-07:
First time I open an issue here. So first of all thanks everybody for all the great job done here :)

I'm a gadgetbridge user for years with a MiBand 2. Now I have my BangleJS I'll try to help to get more reliable information.

Here I'm trying to focus on activity reporting when we enable hrm in the settings.

So we are not in `realtime` and `hrmTimeout` is set to `5`.

In that situation we are activating `setHRM('on')` every 5s which looks very quick to get accuracy.  The `hrmTimeout` is also used to try getting accuracy in the 5 `Bangle.on('HRM')` to come. (does it trigger every second?). If we don't get a good confidence in the 5 times we then send a non accurate value (confidence > 20). And this is triggered every 3min based on default `activityInterval`.

Based on what I've seen confidence has to be near to 80% or more to represent a correct value. I'm wondering if it is interesting in non realtime situation. I'd prefer having a value every 30s even 1min which represents a correct value rather than many inaccurate value.

Is it possible to increase `hrmTimeout` to `30` for example and maybe tries to get two value with 80% of confidence and doing an average? And in all case if accuracy is not good not sending that information?

Almost same about steps which are crazy on my side: 120k steps in a day. Why don't use the counter provided by an existing widget like activepedom?

@rigrig @gfwilliams

gfwilliams, 2021-04-08:
Hi! There's actually a bit of discussion on this on the forum at the moment: http://forum.espruino.com/conversations/362096/#comment15923346

But I think realistically a lot of the issues come from the algorithms on the Bangle itself - the HRM confidence value can be too high in some cases, and the pedometer over-records as well. It's something I have plans to get fixed soon.

The HRM event happens every 5 seconds, so the test will return a value every 25s (close to what you wanted). Do you think it should maybe just not return a value if it doesn't get an accurate one?

For pedometer, the idea is to fix the step recording properly (active pedometer seems to have some issues too) and to then strip all the extra logic out of active pedometer so it just uses the bare values reported by Bangle.js.

valvin1, 2021-04-08:
It looks I misunderstood the code :) so turning `setHRM('on')` returns only one `bangle.on('HRM')`? 

I remember having many records on my gadgetbridge activity like every 5s. But now I can't reproduce. I'm wondering if it wasn't linked to HRM recorder which was also running.

> Do you think it should maybe just not return a value if it doesn't get an accurate one?

Yeap I think sending a wrong information is not good. For example  inaccurate values are 40 or 200 if they are send the graph is weird. We don't see it because graph is tricking to have a nice trace. Then it depends on how many times we hit 5 times and not confident value.

About the discussion it's also me :)

valvin1, 2021-04-08:
I think I've understood the code now :) It's easier when trying on the IDE.

If I've well understood:

Reported value is every 3min by default.

If HRM is enabled in settings. Then HRM is on at least 5s and max 25s. if in the fourth result confidence is greater than 80% it is recorded. If the fifth records has a confidence greater than 20% it is sent else we don't send the hrm information.

My suggestion would be testing HRM longer than 25s if no confident value is found before. And add a new var `hrmAttempts = 10` and use it instead of `hrmTimeout` to stop HRM. Sending `-1` if no confident HRM (> 80%) is found during this timelapse.

In a second time based on what I see I think to get this more reliable we should have two confident value successively and without a too big delta ( < 20)

I can try to implement this if you think it could be a good idea.

gfwilliams, 2021-04-09:
Thanks - the HRM data and confidence is calculated on a block of data though, so if confidence it working right then it should be fine to work based on just that block.

For you, right now, this looks promising - but I'd prefer to fix this properly in the firmware rather than adding hacks in JS to work around it. The code that works out the confidence is at https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L2649-L2651

If you're interested I think what would really help is to capture your raw heart rate data (there's an app for it) and then we could run the bangle.js C code over it and see why it's not producing good confidence values - or we could even try out some different algorithms to see if there's anything else that behaves better.

My gut feeling is maybe it makes sense to go for a much more simple HRM calculator based on peak-finding and to then use the accelerometer to discount any HRM data that occurs during fast movement

valvin1, 2021-04-09:
Yeap I would also prefer that sent information with a high confidence is reliable :)

I'll try to capture some data with different use case. I'm not sure I'll be able to test C things but I'll give a try.

In my experiments non reliable information occurs when watch moves even a little which is not very convenient. I've added other controls to get a more reliable data by removing value bellow 50 and equals to 200.

In case it interests other I've put the code here: https://github.com/valvin1/BangleApps/tree/gbridge/hrm-accuracy

valvin1, 2021-04-10:
about the pedometer and activity in gadget bridge, i've implemented a possibility to get information from widget. As I've found a good settings now for activepedometer I have corresponding information in gadget bridge and it detects activity. For those interested it is in the branch mentioned before.

With this version HRM data are ok but unfortunately it misses many data as there are considered inaccurate.

I have done one export of HRM during 5min ([hrm_log.csv](https://github.com/espruino/BangleApps/files/6289815/hrm_log.csv)). This export corresponds to me doing activity (here cooking).

I've use this [script](https://github.com/jabituyaben/BangleJS-HRM-Signal-Processing) mentionned in the app and I get the following result:

```
{'mean_nni': 766.8333333333334, 'sdnn': 63.1657834679292, 'sdsd': 82.39572931427892, 'nni_50': 91, 'pnni_50': 50.83798882681564, 'nni_20': 129, 'pnni_20': 72.06703910614524, 'rmssd': 82.39603233925403, 'median_nni': 760.0, 'range_nni': 225, 'cvsd': 0.10744972702358709, 'cvnni': 0.08237224533961643, 'mean_hr': 78.77216147421665, 'max_hr': 90.9090909090909, 'min_hr': 67.79661016949153, 'std_hr': 6.451098535130338}
```

`*_hr` values looks correct. Hope it could helps. Let me know what kind of other data you need. I've thought of those:

* no activity (clean data)
* during a walk
* during sport
* during computing :)

gfwilliams, 2021-04-12:
Thanks! That's really handy - interestingly that script would seem not to use the filtering method that Bangle.js uses (but just a simple peak finder after some filtering).

I'm not quite sure when I'll get time to look at this in detail, but I'm sure we could do a lot better than we currently are on bangle.js

gfwilliams, 2022-02-03:
I'm going to close this now, as the step counter is vastly improved, as is the HRM. I believe the activity recording is a lot better

closed 2022-02-03 by gfwilliams