espruino/BangleApps #718: gadget bridge - offline activity

valvin1, 2021-04-11:
This is a feature request which is I think a bit difficult to implement. It is related to #716 

At the moment, activity is sent only when BT connection between smartphone and bangle is established. As step counter is cumulative it doesn't work if during a period the connection wasn't established.

I think having the same behaviour as MiBand2 with Gadget Bridge could be interesting. MiBand2 stores data locally and regularly GB get the data with possibility to let data locally or delete it. That way only GB get data and locally on the bangle activity are stored doesn't matter if link is established or not.

So there is two feature : 

* change how activity is handle in banglejs gadget bridge. it should store every 3min (or another period) in file values of steps, hrm timestamped. It should be interesting having other data like sleep detection and movement score.
* change in GB android app which scrapes regularly the banglejs and requests data it has not yet collected. I think there is already an interface existing for this. Maybe it is related to this not yet implemented function [here](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L484)

gfwilliams, 2021-04-12:
Yes, this is definitely something that could be included.

I'd be interested to know what proportion of users would actually end up using this - I guess many have the watch in close proximity to their phone most of the time anyway.

I wonder if there's anything to be gained from having a more generic storage system for HRM/pedometer/other data that all different apps could access.

valvin1, 2021-04-12:
On my side i don't have always my phone with me and even on short distance I can have disconnection. I've also noticed it is not always well detected don't know if it is on bangleJS on gadget bridge side.

Tonight GB has 1564 steps vs 4164 on my bangleJS ;-)

I think having a global file with timestamped values of HRM / pedometer will be interesting not only for gadget bridge. At least for clocks but I can imagine it could be useful for other apps.

gfwilliams, 2022-02-03:
Update: we now have that global file of info via the `health` app. I guess we just need to integrate the 'activity fetching' from Gadgetbridge

Moini, 2023-03-23:
Hi - yes, as a new owner of a Bangle.js 2 and owner of a no longer working Amazfit Cor, I'm missing this. 

There was a button in Gadgetbridge that allowed me to sync the data for the other watch. 

I usually do not walk around with my mobile's bluetooth on, but turn everything off that connects outwards for as long as I do not need it. It's too old an insecure for that... Also, as the Bangle.js does not seem to have a way to prevent others from connecting to it, I do not even want to keep it in a connectable state continuously. Battery saving is a plus. And most of the time, I do not carry my phone on my body, while still working and walking around. It's just not feasible - I don't always have pockets.

But I would like to be able to see the sleep log as well as properly working heartbeat sensor data (another issue...) as well as steps and activity data in Gadgetbridge again. They are all recorded, so even if they're currently not very accurate (at least the heartrate), it would make the watch so much more useful.

gfwilliams, 2023-03-24:
Hi,

This is mentioned in https://forum.espruino.com/conversations/384986/#comment16917728 too.

Just to explain where we are with this: The `health` app stores HRM data every 10 minutes on the Bangle, and if connected by Gadgetbridge it sends that data too.

But if not connected, that data is never sent to Gadgetbridge. As I understand it, it'd be a reasonably easy job to:

* Make Gadgetbridge's `"act"` handler take an optional timestamp argument: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L552
* Add a handler for `onFetchRecordedData` that asked Bangle.js to send the data from the 'health' app back: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L1193-L1194

However the issue really is knowing how much data to send - you can't just send data from every 10 minutes for the last year every time you're asked - so I guess somehow Gadgetbridge is going to have to know what kind of 'hole' it has in the data that's been recorded, and then request just that?

That's the bit I'm unsure about... Perhaps at connection time we just get the Bangle to automatically send the data it collected since the last time it was disconnected?

Moini, 2023-03-24:
Hi Gordon, thank you for replying.

I think I'd hand the responsibility about what data is needed to Gadgetbridge. That way, if the data flow was interrupted, or someone uses two smartphones with two Gadgetbridges, it will still get the missing data. 

(btw. for me, this night, almost nothing has been reported for most of the time... a single heart rate reading is what I found in Gadgetbridge this morning, despite having had Bluetooth turned on for testing - activity data is never reported, and steps are only available after the day has ended, I think?)

Moini, 2023-03-25:
Thanks, yes - I had not disabled the battery optimization. It had probably worked the first time, because the Gadgetbridge bangle.js app was in the foreground. 

Tonight, after disabling the optimization, only very few measurements made it through to the phone - I cannot tell whether they were too imprecise to be accepted, or what happened. It is set to 'every 3 minutes' on the watch. There is a 3.5 hour gap with no measurements at all, but the line is drawn there, so I guess connection was established. Maybe that's a matter of arm posture during sleep...

I'd request the data added since the last valid measurement from Gadgetbridge. With an option to either keep it on the watch or to delete it after successful transmission.

Moini, 2023-03-25:
(and maybe with a limit to 'last 10 days' or something, so there's no overwhelming flood of data... ?)

gfwilliams, 2023-03-27:
Ok, thanks - are the HRM readings looking ok in the 'health' app on the watch?

Just out of interest, is there a reason this came up recently? Because Gadgetbridge has been working this way with Bangle.js for probably a year now and nobody seemed too bothered, but in the last week I've suddenly had several complaints about it (between GitHub, the forum and Twitter).

Moini, 2023-03-27:
As far as I know, the health app only shows aggregated data (per hour, per day), so I cannot tell how many readings there were... :-( , and I can't go back a whole day, to compare the 'holes' in the data. It *does* look like there may be more measurements available on the watch than there are listed in Gadgetbridge bangle.js, but I cannot tell with 100% certainty.

About the second question - no idea. I only got my watch on the 17th, so this is why I've come here at this time, to see whether it can be improved in some way. Maybe I was unlucky, and it worked fine with previous versions - I cannot tell. After I got it, I updated to the 1.17(? not sure about the number, web interface doesn't show current when not connected) firmware right away.

digitalcircuit, 2023-06-29:
@gfwilliams
> Just out of interest, is there a reason this came up recently? […]

It took me a few months to discover this issue, around the time I actually started aiming for 10k steps a day.

It's not obvious what's happening, and only when I went on an entire walk with my phone without realizing Bluetooth randomly disconnected and didn't reconnect that I even knew something was up (afterwards I switched from `widbt` to `widminbt` to make it more obvious).  I had previously assumed that this simply synced, especially since tapping on the Gadgetbridge widget said "Fetching activity data…"

Now I am very interested in this feature so I can choose to leave my phone behind (e.g. to avoid distractions) and not worry about missing steps. :slightly_smiling_face:

m0n5t3r, 2023-07-17:
for what it's worth, my phone receives between 1/3 and half of the steps from the watch, and this after I did all things listed somewhere in the documentation to the gadgetbridge app... this was never a problem with banglejs v1 and the same phone / OS (graphene)

periodic sync would also do wonders for battery life

joserebelo, 2023-07-18:
I started looking into this, since I noticed that there was no sync method on Gadgetbridge (and actually forgot to check the issues.. was too keen to start coding :)

#2889 introduces a small API to fetch missing data, from the last timestamp that Gadgetbridge saw until the current time, still WIP.

Moini, 2023-07-19:
> that there was no sync method on Gadgetbridge

That is unexpected, as there was a sync button in Gadgetbridge for my previous watch, an Amazfit Cor.

thyttan, 2023-07-19:
Well, generally there are sync methods for most watches/bands in Gadgetbridge. It's just that it's not implemented for bangles.

joserebelo, 2023-07-19:
Yes, sorry - by "there was no sync method on Gadgetbridge" I meant that it's not implemented for the Bangle :)

Moini, 2023-07-19:
Ah, thanks! I was afraid of someone starting to reimplement the wheel in its entirety.

Moini, 2023-08-09:
@joserebelo Yay :D So cool! Can't wait to try it out. 

Where should bug reports go, in case something's not going well?

joserebelo, 2023-08-09:
Probably on the Gadgetbridge repository, since the implementation is Gadgetbridge-specific, and bugs will most likely need to be fixed there anyway.

If you open any here, tag me, since I don't actively monitor issues in this repository :)

Moini, 2023-08-09:
@joserebelo Thanks!

PR #2889 (2023-08-09): Implement API for activity fetching
Add a simple API for ativify fetching from Gadgetbridge. The API receives the start timestamp, and returns activity up until the current timestamp.

PR on Gadgetbridge: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3212

Not incredibly happy with the code, but I am trying to avoid iterating through entire months needlessly, since sync might happen periodically.

Still very much a WIP.

Fixes #718