espruino/BangleApps #750: lauchner buzzes on every key press

neshanjo, 2021-05-30:
Hi,

some of the recent updates introduced "buzzing" on every key press in the launcher.
https://github.com/espruino/BangleApps/commit/6d1ac3b7c62dd92aabf029c38a16935f23b609fc#diff-994bb4ae20c0cb9d61cc11c4841860d2f92ccda47bfd23b69ee549b84823ddd1

For me personally, the haptic feedback of the buttons is sufficient, so I did a first try of implementing a new setting "Menu Vibration" which configures the menu vibration intensity (see https://github.com/neshanjo/BangleApps/commit/70f9fd13d1c10f74b35f29723dd6f171f9c16797). However, this is not working. Probably because I did not 100% understand how the bootupdate.js works. Before putting more effort into this, I'd want to ask if the buzzing is really needed (or can maybe reduced to an intensity <= 75 which is - at least on my bangle - a better value). Or else, if there is some other way of making it configurable.

Kind regards,
neshanjo

gfwilliams, 2021-06-07:
Sorry for the delay - thanks, yes, that should definitely be an option. And probably as you say it's not needed and could be at minimum off by default.

You can just turn the buzzing off in options and that'll fix it, but obviously it stops all buzzing.

It seems like what you've done should work, but it's probably almost better to do:

```
${s.vibrateMenu?`try{Bangle.buzz(20);}catch(e){}`:``}
```

instead. Having said that, the code is kindof a 'polyfill', so if you're running a 'cutting edge' build then I think the `setUI` function is built in and that code doesn't get called.

neshanjo, 2021-06-07:
> You can just turn the buzzing off in options and that'll fix it, but obviously it stops all buzzing.
That's not an option, because I'm using apps where I need the vibration (e.g. the metronome app).

> Having said that, the code is kindof a 'polyfill', so if you're running a 'cutting edge' build then I think the setUI function is built in and that code doesn't get called.
Ah, ok, I missed that (though it's well documented in the code 🤦 ). That explains why it had no effect on my device!

Changing the firmware goes beyond what I can do in my spare time, so maybe you can accept this as a feature request. It would also be ok for me, to remove the buzz completely at this point as a simple solution.

gfwilliams, 2021-06-08:
Thanks - yes, I'll remove the buzz.

I guess the nice thing with this approach is we can easily create an app that can be installed which replaces `setUI` and adds buzz later if someone wants it

closed 2021-06-08 by gfwilliams