espruino/BangleApps #752: gpsrec lost part of the trip

Weiming-Hu, 2021-06-04:
I have been using GPS Recorder (v0.19) for a while now. I found out that, for my recent two trips, gpsrec lost part of the trip recordings. For example, I went out for a run for an hour but it seems the first 20 minutes or so was missing in the file.

I set the frequency to be every 15 seconds. And I have enough space on my watch. I'm not sure why this is happening. Did anybody has similar experiences with gpsrec? Any ideas on how to debug?

Thank you

Weiming-Hu, 2021-06-05:
OK. I think I have figured this out by reading the source code [here](https://github.com/espruino/BangleApps/blob/master/apps/gpsrec/widget.js#L28).

Both my trips were missing the starting part of the recording, about 10 minutes for the first trip and probably more than 20 minutes for the second trip. Once the trip started recording, all points afterward seem to be successfully recorded.

`gpsrec` won't write any information to the file if there is no GPS fix and GPS location fix can take a while during a cold start. As a result, there could be a lag from the time GPS is turned on and when the trip is actually recorded. Once it starts recording, there seems to be no problem.

And I just notice that the little GPS badge icon shows up differently when you have or don't have a fix. So I guess I can pay more attention to that next time.

I'm closing this since it is not a bug issue but really just a user habit.

closed 2021-06-05 by Weiming-Hu

gfwilliams, 2021-06-07:
Ok, thanks! Yes, I think it's likely the time to get a fix that's an issue. If you install 'AGPS' in the few days before you go out, that can really help with the time to fix.

However maybe we should add some docs to GPSrec to help explain? The icon is a bit hard to see.

Weiming-Hu, 2021-06-11:
Just created a PR [here](https://github.com/espruino/BangleApps/pull/760)