espruino/BangleApps #761: distance difference between banglerun and gpsrec

bernhardreiter, 2021-06-15:
Tried gpsrec and banglerun a few times, and noticed a significant difference in the distance between those two.
What could be the cause?

Example:
 * Start gpsrec on a new track, (recording every 10 second), go back.
 * Start banglerun
 * Run 
 * Look at banglrun (recorded 10.35 km), then press button 3 to stop and go back to the main page
 * Stop gpsrec, look at the track and plot the distance (8.9km), within the same time.

As both claim to use the GPS, the difference cannot be explained by gpsrec starting the recording later.
Any hints, ideas?

gfwilliams, 2021-06-15:
That is an odd one - however while banglerun will record the data less often to flash, it will look at GPS coordinates to log the distance every second, so if there is any randomness to the value that it receives from the GPS, that will add up over time.

Imagine if you ran quickly in a zig-zag. Obviously in the case, if you work out distance from a fix once a second it records more than if you do it once every 10 seconds.

We did originally have a Kalman filter in Banglerun which made a nice job of this, but some changes broke it and I couldn't easily fix it so it got removed. I think it's something that longer-term I'd like to add inside Bangle.js to provide more nicely filtered GPS data

bernhardreiter, 2021-06-15:
The difference seems to be too big (more than 10%) to account for cutting edges between 1 second and 10 second measuring interval, (because I did not run so many corners). However my next analysis steps:
* Try to download, visualize and measure the gpsrec tracks in a GIS
* Try with smaller measuring interval of gpsrec

gfwilliams, 2021-06-15:
Yes - you can also download the banglerun tracks too, but they should contain basically identical data

bernhardreiter, 2021-06-25:
Used  https://f-droid.org/en/packages/mobi.maptrek/ to view a track that has been recorded by gpsrec and banglerun. For the mentioned track there was only a difference of 62 meters, which is fine with different recording intervall and potentially a few seconds difference in stopping both applications. gpsrec had 8.971 and banglerun had 9.033. But banglerun was showing ~10.35 on the display, so my guess is that something is off with the live updated summary of the track distance in banglerun.

(BTW: would be a cool debugging feature if gpsrec could also display the tracks of banglerun).

gfwilliams, 2021-06-25:
I think the issue is as I said above - I guess banglerun is recording every 3 seconds or something, gpsrec is recording every 10 and gets less distance.

However BangleRun's display is using the GPS data that occurs every second to work out a distance, which means it records an even higher distance.

The Kalman filter would be the solution here...

bernhardreiter, 2021-06-25:
The distance as calculated by Trekarta (former MapTrek) for the track from banglerun was shown as 9.033km, but banglerun itself showed much more about 10.35km (this is from my memory as I when I stop banglerun, the display is gone of course, but I am sure.) The same track recorded with 10 seconds with gpsrec and shown and calculated by trekarta was 8.971km long, so within trekarta the difference between the two intervals is 62 meters. Therefor my conclusion that the problem for the live summary of banglerun must be a different one, it is not the interval difference.  

Another experiment a few days ago, I've changed the interval on gpsrec to 1 second (because I've thought that would be what banglerun uses), again there is a large difference. However I could not compare this in trekarta yet, because downloading did not work #766 :) )

gfwilliams, 2021-06-25:
> Therefor my conclusion that the problem for the live summary of banglerun must be a different one, it is not the interval difference.

I don't really understand what you mean here.

But I think that when you do get the 1 sample/second data off the Bangle, you'll find that if you sum the distances between fix points it will exactly match what BangleRun reported

bernhardreiter, 2021-06-25:
> when you do get the 1 sample/second data off the Bangle, you'll find that if you sum the distances between fix points it will exactly match what BangleRun reported

That is not what I am seeing. 

I saw BangleRun reporting 10.35 (memory), and Trekarta 9.033 (via gpx download from banglerun) on the same track. 

This would also explain the difference of BangleRun reporting to the gpsrec over the same session. gpsrec session displayed on the watch and on trekarta (gpx download) matches up.

gfwilliams, 2021-06-25:
But what data was Trekarta using? Wasn't it data only recorded every 10 sec or 3 sec, not every 1 second?

bernhardreiter, 2021-06-25:
Let me call this one session where I took the watch to physically run roughly 9km session A.

In session A I have started both first gpsrec, then banglerun, using the default values. I've looked at banglerun, memoriesed the distance it displayed, stopped first banglerun, then gpsrec at the end of the session. A has been a few days ago. Later I've downloaded both recordings as gpx data files, both covering A.

Then I've displayed them both with Trekarta, I could see that banglerun used more datapoints. So the banglerun track has whatever banglerun uses as interval (given from what you have told me, probably 3 secs) and gpsrec had 10 secs.

gfwilliams, 2022-02-03:
Closing this now as we have the `Run` app and `Recorder`. There's some discussion on the forum but I think everyone seems happy with the distance calcs now: http://forum.espruino.com/conversations/371899

closed 2022-02-03 by gfwilliams

bernhardreiter, 2022-02-03:
> Closing this now as we have the Run app and Recorder.

Okay, if this is a no-fix because banglerun is outdated.

In my Run and Recorder apps, I've experienced the same defect of the Run calculation, just added this to the discussion you've mentioned. http://forum.espruino.com/comments/16382715/

gfwilliams, 2022-02-03:
Ok, lets discuss on there... I'm deprecating BangleRun and will probably get rid of it at some point soon, but if we can find the same issues with the Run app then obviously that's an issue