espruino/BangleApps #764: Can't seem to "Get out of" the GPS Time App

mattpackwood, 2021-06-24:
Hi,

I seem to be "stuck" at this screen, how do I get out of the app without resetting the watch?

Matt

gfwilliams, 2021-06-25:
You just long-press BTN3 (bottom right)? It's the way you normally get out of apps unless they have specifically dedicated a button to exiting.

Or does that not work for you?

mattpackwood, 2021-07-14:
Yep, BTN3 gets me out, although the screen says BTN2

[image]

gfwilliams, 2021-07-16:
Hi - yeah, the 'Set' by BTN2 means it'll set the system time to the GPS time you see there.

But I guess it does make sense to add another label and allow short-press to exit. I'll be making some changes to get this working nicely with other devices so I'll look at adding the functionality then

mattpackwood, 2021-10-21:
Many thanks!