espruino/BangleApps #766: gpsrec could not download tracks after upgrade 0.19 -> 0.21

bernhardreiter, 2021-06-25:
There are a few tracks on the watch, which I can view and could download.

Yesterday I did an update from 0.19 to 0.21 (it was the only update) and pressed button 3 for reload.
Now downloading tracks from the web interface did not work anymore, there were non found.

Expectation: Tracks should be downloadable after and upgrade.

gfwilliams, 2021-06-25:
Can you still view the tracks on the watch itself? Or is it totally gone now?

bernhardreiter, 2021-06-25:
The tracks can still be viewed on the watch.

gfwilliams, 2021-06-25:
Ok, thanks. The actual app shouldn't have any effect on whether you can get tracks off or not though, since the App Loader talks directly to the watch, avoiding the apps.

Do you think you could open Chrome Developer Console after trying to view tracks with the Bangle App Loader, and then paste the contents of it in this issue?

bernhardreiter, 2021-06-25:
It is the Android Chrome browser, I'll see later or tomorrow if there is a development console I can find.

gfwilliams, 2021-06-25:
I notice you just mentioned in issue #761 that you've never actually been able to download the track? 

I assumed that you had previously downloaded it and now can't - but are you saying you recorded a detailed track at 1 sample/sec and then you can't download it? How long did you record for? It might give me an idea of file size.

And which Bangle.js firmware version are you using? It should say in the about screen.

bernhardreiter, 2021-06-25:
With 0.19 I could download tracks, also some with 1 secs measurement, only after the update yesterday, I couldn't. Your idea is good, though: I haven't tried to download the very long one before, so it maybe unrelated to the update, but related to the size of one track.

The long track is 5947 records long.I can view it on the watch (plotting the map takes a few seconds ;) )

gfwilliams, 2021-06-25:
> only after the update yesterday, I couldn't.

I think that might be coincidence. `interface.html` - the thing that actually reads the data off the watch - was last updated **3 months ago**. So I don't believe anything has changed there.

We have had issues in the past with it timing out when it takes too long to receive data on tracks though, although I thought I had fixed that.

I've just made a change that might improve matters for you. Maybe try the development version of the app loader at https://espruino.github.io/BangleApps/#gpsrec and see if that helps?

Connecting with the Web IDE and issuing `require("Storage").compact()` would probably speed the watch up and might fix it too, but make sure you're on the 2v09 firmware first

bernhardreiter, 2021-06-29:
>  And which Bangle.js firmware version are you using?

I think I am on the latest one (at least I've updated everything), but I'll check.

> Maybe try the development version of the app loader

I will, but it may take a few days until I'll have the opportunity.

> issuing require("Storage").compact()

You mean like on the development console of the browser?

gfwilliams, 2021-06-29:
> You mean like on the development console of the browser?

No, that'd be in the left-hand side of the Web IDE at https://www.espruino.com/ide

`compact` basically 'defrags' the flash memory. It removes any deleted/modified files from flash and moves them all back to the start - it can really speed up accesses

bernhardreiter, 2021-07-05:
Firmware checked (boy that small font is hard to read): still at 2v08, so I look into flashing this next.

Got a bluetooth dongle finally working on GNU/Linux with chromium, to have a development console. Tried again: Now I got offerend to download tracks with gpsrec. Thus no error message in console. Have to try this again with the other android device where downloading was failing before. (After that last try I had deleted two banglejsrun tracks.)

bernhardreiter, 2021-07-05:
Today it also works for the android device where it was failing before.
So the testcase is gone. I don't know why, but without a testcase, I close this issue for now.

What could have changed:
* deleted two banglerun tracks (but no gpsrec tracks), downloaded all of them
* used chromium on a desktop once
* maybe something on the  website had changed

closed 2021-07-05 by bernhardreiter

gfwilliams, 2021-07-05:
Great! I mentioned in https://github.com/espruino/BangleApps/issues/766#issuecomment-868383195 - I made a tweak to the code, so likely that is what fixed it for you

bernhardreiter, 2021-07-06:
> >  I've just made a change that might improve matters for you. Maybe try the development version of the app loader 

>  I made a tweak to the code, so likely that is what fixed it for you

Note that I've used the regular version at https://banglejs.com/apps/. If the code made it there from the development version meanwhile, then it could have been used. 

Thanks in any case. ;)