espruino/BangleApps #768: Bad color value formatting on battery and bluetooth widgets

smartlitchi, 2021-07-09:
After updating Battery Level Widget to v0.07 and Bluetooth Widget to v0.05, they were unable to load because of this error : 
``Uncaught Error: If Color is a String, it must be of the form '#rrggbb'``.

gfwilliams, 2021-07-09:
Hi. Thanks for this, but which Espruino firmware are you on?

I believe you're probably on a pretty old firmware - the app loader should really have warned you. Did it not?

On recent firmwares that new form of color value is totally fine.

gfwilliams, 2021-07-09:
Just checked - that was stupid, sorry. The app loader only warns if you're below version 2v08, but the `#rgb` thing came in in 2v09 which has been around for a few months now.

I've just updated the app loader so it'll warn folks to update.

Thanks for letting me know and for the PR, but I think in this case it's worth just updating the Bangle.js firmware to the latest version (as the `#rgb` form is used in a whole bunch of apps now)

gfwilliams, 2021-07-09:
'Fixed' with https://github.com/espruino/BangleApps/commit/6670d517243934e809fe462c342fcc4885bdf7d6

closed 2021-07-09 by gfwilliams

smartlitchi, 2021-07-09:
Yes I was on 2v08 (just received the watch today) so it explains the issue. Thanks for pointing me to the solution. After updating the firmware, everything worked just fine