espruino/BangleApps #781: Trying to connect to a password-protected BLe device

nakeze, 2021-07-29:
Hello, I just got a bangle js today, it's dope ! Javascript and the web IDE makes it easy to prototype. I started to work on a project where I have to receive and send data to a peripheral BLe device.
I got no problem to find my device using the code from the BLe scanner, but unfortunately, i didn't find yet how to connect to it (also how to be notified of the chars modifications).
I have to connect to a device that needs to be visible (so no white list really possible), but still protected (a password that is hard written seems a good solution). 
I already develop a react native app that is able to communicate, the password part is at the pairing stage (if I am not wrong). 

I checked the NRF Espruino library and I understood the options for connect have no mention of password, also found it is possible to create a peripheral with a password. 

I will continue to check if there is not an already made app using this feature.
Can anyone explain me how to communicate from a central point of view using espruino or send a link explaining it ?

Sincerely,
Nicolas

gfwilliams, 2021-07-29:
Hi - please can you ask questions like this on the forum in future? http://forum.espruino.com/microcosms/1424/

So to be clear - you want:

* Bangle.js as a central device to connect to a peripheral
* That peripheral has a passcode

That sound right?

If so, I think what you want is to add a handler for the `BluetoothDevice.on('passkeyRequest', function() { ... });` event after you got the device, but before you connect. In it, you call `BluetoothDevice.sendPasskey("123456")`

nakeze, 2021-07-29:
Thanks for your quick answer ! It sounds right. 
I will definitely use the forum next time. I understand now that i need to handle the resolve object of the NRF connect : BluetoothRemoteGATTServer classes. I will try to fix my issue with what you said and will open a issue on the forum if needed.

closed 2021-07-29 by nakeze