espruino/BangleApps #782: sma q3 setting menu style for lefties

MaBecker, 2021-08-10:
As lefti I scroll with my left hand through the settings menu and my finger hides the text.

Can we have a left hand menu style with  text that is right aligned?

gfwilliams, 2021-08-12:
You mean with values on the left and the text on the right?

That's probably not something I'd build in but it should be trivial to add as an app that replaces the built-in `E.showMenu`. I'm happy to help with that

MaBecker, 2021-08-12:
After using it for more setting, I really like to have a larger font because of the size of my finger can’t scroll between lines easy …..

gfwilliams, 2021-08-12:
The menu is designed so that you just move your finger up and down, rather than having to tap on a specific menu item for exactly that reason (I found even 50% larger didn't really help). The sensitivity of that scrolling could easily be less (or even configurable) though?

MaBecker, 2021-08-12:
Oh yes, change the sensitivity for those large menues would really help.

gfwilliams, 2022-03-24:
just to add this may be fixed by https://github.com/espruino/BangleApps/issues/1040 I guess

gfwilliams, 2024-05-02:
No response - lets assume that #1040 did fix it

closed 2024-05-02 by gfwilliams