espruino/BangleApps #786: Bangle2 (SMA Q3) issue: Device no longer visible after BT disconnect

yngv27, 2021-08-11:
I had the B2 connected to the IDE running well, I walked away from the computer, which disconnected the BT; when I tried to reconnect, the new Bangle was not on the BT list, even after turning off/on BT on computer and refreshing IDE page. Device showed up again after long button press (default clock)

gfwilliams, 2021-08-12:
I just tried it exactly that here and I don't see the problem at all. It shouldn't really behave any differently to the existing Bangle.js 1.

Did you have Gadgetbridge running on your phone, or some other device that might have started connecting to the Bangle when the IDE disconnected?

yngv27, 2021-08-12:
OK, just tried it again today, no problem. Weird. If it happens again repeatably, I'll log it back in.

closed 2021-08-12 by yngv27