espruino/BangleApps #819: Layout

nebbishhacker, 2021-09-22:


gfwilliams, 2021-09-23:
Thanks! That's interesting. I thought pad got added to the required width, but maybe not... I'm starting to wonder if it's worth implementing a test harness now (JS + expected output). It might also serve as a useful reference.

closed 2021-09-23 by gfwilliams

nebbishhacker, 2021-09-23:
Here's another test case that still exhibits issues.

```
const Layout = require("Layout");

g.clear();
var layout = new Layout({type: "v", c: [
  {type: "h", pad: 20, c: [
    {type: "txt", font: "10%", label: "abcd"},
    {fillx: 1},
    {type: "txt", font: "10%", label: "1234"},
  ]},
]});
layout.render();
layout.debug();
```

Expected:

[image]

Actual:

[image]

commit 0109a811 (2021-09-23): Ensure padding works correctly (fix #819), also add Layout test harness basics