espruino/BangleApps #820: Layout: unable to control `fillx`/`filly` of parent layouts

nebbishhacker, 2021-09-22:
If `filly` is set on a child element, it seems to automatically get set for its parent layout as well. This makes some layouts difficult to achieve, and makes nested layouts a bit harder to reason about.

My preference would for `fillx`/`filly` to always default to `0` for parent layouts regardless of what their children are doing. Failing that, I would like to at least be able to manually override it.

---

```
const Layout = require("Layout");

g.clear();
var layout = new Layout({type:"h", filly: 0, c: [
  {type: "txt", font: "50%", label: "A"},
  {type:"v", c: [
    {type: "txt", font: "10%", label: "B"},
    {filly: 1},
    {type: "txt", font: "10%", label: "C"},
  ]},
]});
layout.render();
layout.debug();
```

Expected result:

[image]

Actual result:

[image]

---

I ran into this while trying to replicate the layout of the weather app.

gfwilliams, 2021-09-23:
Ahh, good point! Yes, that seems fair.

The reason it got added was for the acceleration logger: https://github.com/espruino/BangleApps/blob/master/apps/accellog/app.js#L100

Basically I had a simple vertical layout but wanted one heading to go right across so the background was all filled.

In your case, I guess you'd actually need to explicitly set fill on the vertical group, but not on the horizontal?

nebbishhacker, 2021-09-23:
> In your case, I guess you'd actually need to explicitly set fill on the vertical group, but not on the horizontal?

Yes, in this case the desired end result would be for the vertical group to have a filly of 1 and the horizontal group to have a filly of 0.

Thinking about it more, I'd say the current behavior makes sense in most cases (and, indeed, matches with my intuition more often than I realized) - so all I really want is to be able to override it by explicitly setting the parent fill.

commit 807ac743 (2021-09-25): Layout: only set fillxy/y if null

Fixes #820
closed 2021-09-25 by gfwilliams