espruino/BangleApps #892: 12 hr clock setting doesn't work

eyecreate, 2021-11-18:
I changed my locale to en_US and went to locale menu and changed clock to 12hr. Anton clock should be able to handle since it asks locale module for time, but it still shows up as 24hr time.

gfwilliams, 2021-11-19:
Thanks - I'll check on that. Hopefully it's just an update to the locale module

jg76379, 2021-11-19:
I took a look at the locales module and it doesn't looks like it is reading setting.json to get the 12 hour time setting. 

It seems like it should be doing something like the following to get the setting:

```
const storage = require('Storage');
var settings = storage.readJSON('setting.json', 1);
var time12Hr = settings['12hour'];
```

And then modifying the time format, but i'm not sure how to implement this because of the way that the locale module is setup. Since that code within `locale.html` is running in the browser to generate the actual locale code that is sent to the device.  Where would you check the settings to adjust the time format?

Izaic, 2021-11-23:
Hopefully it's an easy fix

closed 2021-11-29 by gfwilliams

JayAndCatchFire, 2021-11-30:
This seems to still be an issue.  Maybe because the build failed?

eyecreate, 2021-11-30:
I'm running build 2v10.234 and it's working. Are you running a different build or have you tried a factory reset?

JayAndCatchFire, 2021-11-30:
Ah, good call.  I'm on 2v10.219.  I didn't realize this update required a firmware update.

JayAndCatchFire, 2021-11-30:
After I updated the firmware, it works as expected.  Thank you!

PR #952 (2021-11-29): locale: Apply 12hour setting to time
Fixes #892

I figured it would be a good idea to not read+parse the settings file every time something wants to draw the time, so we remember it after the first call.
commit 6220a88d (2021-11-29): locale: Apply 12hour setting to time

Fixes #892