espruino/BangleApps #896: [BangleJS 2] Q alarm and timer does not load on alarm/timer activation when another app is open

Aracnoss, 2021-11-19:
Q Alarm and timer fails to load when a timer/alarm goes off while in another app and will just remain in the previous app with the loading text in the middle.

I had this happen when a notification came through from android integration and I hadn't closed it before the alarm went off.

Aracnoss, 2021-11-19:
Sorry forgot to mention that full screen notifications is also installed.

gfwilliams, 2021-11-25:
@qucchia any ideas?

qucchia, 2021-12-07:
I think sometimes the alarm schedules itself just a little too early, enough for there to be no active alarm when it looks for them. I'll see what I can do.

thyttan, 2022-11-28:
Candidate for being closed - leave a comment if you object!

Rationale:
I think this is maybe fixed with [this commit](https://github.com/espruino/BangleApps/commit/426ba76ad4012f14072d1694d36510977346665d)

gfwilliams, 2022-11-30:
Thanks! Yes, I think also since we moved to using the `sched` library a lot of these kind of alarm issues have disappeared since everyone is using the same tested code

closed 2022-11-30 by gfwilliams