espruino/BangleApps #898: [BangleJS 2] Notifications don't appear to be cleared.

Aracnoss, 2021-11-19:
Not sure if this is an issue with Gadgetbridge or BangleJS 2, but notifications on watch don't appear to ever be cleared.

And given they seem to be ordered from top oldest to bottom newest in the menu, that's going to become more of an issue the more notifications that don't get removed.

gfwilliams, 2021-11-19:
This is using the `Gadgetbridge` app on Bangle.js 2, or the `Android` one I guess?

Maybe the `notify-` event isn't being handled properly? I think there's also an argument that all notifications should be cleared when Gadgetbridge/iOS disconnects. iOS definitely re-sends them at startup.

Also, feel free to report issues, but this is the 3rd one in 30 minutes - given the load I'm currently under with shipping I can't respond to everything so unless someone else wants to jump on these, most will have to wait until the remaining Bangles are out the door.

Aracnoss, 2021-11-19:
@gfwilliams The setup is Gadgetbridge on phone, gadgetbridge and message (with or without the android app) on the BangleJS 2.

No worries, I just like to report them before I forget myself, no rush at all. I might even take a look myself if I get some free time this weekend since I know C/JS, but not sure how likely that'll be.

gfwilliams, 2021-11-22:
Ok, thanks! I believe the 'messages' app won't actually do anything useful without the Android/iOS apps installed - so that could be causing some issues I guess? But I'll look into it here when I get a moment.

gfwilliams, 2021-11-23:
Notifications now get cleared on disconnect, so I think that's ok - but good point about message ordering - I'll tweak that

commit 6aa6bcf1 (2021-11-25): messages 0.06: New messages now go at the start (fix #898)
      Answering true/false now exits the messages app if no new messages
      Back now marks a message as read
      Clicking top-left opens a menu which allows you to delete a message or mark unread
closed 2021-11-25 by gfwilliams