espruino/BangleApps #909: messages app

Didgeridoohan, 2021-11-20:
On my Bangle.js 2 (haven't tested on the v1) I've been experiencing that notifications aren't always cleared from the Messages app when they're cleared from the phone. Can't currently pinpoint why this happens, but it would be nice to be able manually remove notifications from the Messages app.

Didgeridoohan, 2021-11-20:
Apart from this being a feature request, it might be related to #898, although on my device most notifications do clear in the Messages app.

johan-m-o, 2021-11-23:
@gfwilliams Updating to Android integration v0.0.2 and Messages v0.04 removed all those old notifications I had laying around. Nice!

Can also confirm that music controls now work too.

gfwilliams, 2021-11-23:
Great! Hopefully I'll get a few more tweaks in soon too.

commit e8264cc8 (2021-11-25): messages 0.07: Added settings menu with option to choose vibrate pattern and frequency (fix #909)
closed 2021-11-25 by gfwilliams