espruino/BangleApps #922: [Feature Request] messages auto hide after timeout

eyecreate, 2021-11-21:
I get notifications sometimes where I don't address them from my watch. I often want to make sure I can see the time on my watch, but notifications currently only dismiss after you either click a button or hold the side button. It would help a lot to have an option(or default) for messages to disappear after a time so the time (or whatever else was up) is still on screen.

gfwilliams, 2021-11-22:
Thanks - yes, that'd be a nice idea

jeroenpeters1986, 2021-12-06:
That is a great feature request @eyecreate! In the meantime I can recommend the "Digital clock widget" app. This does put a simple digital clock in the widget area whenever something else than the clock is displaying on screen (and there is no fullscreen app active). I use it to my pleasure :-)

jeroenpeters1986, 2021-12-13:
@gfwilliams I think you already implemented this when you released `settings 0.35`

gfwilliams, 2021-12-13:
Thanks - yes, this was fixed as of version 0.09 of messages. Looks like I mentioned the bug number in the commit f848f6977 but forgot to add a '#' so it didn't close this bug!

closed 2021-12-13 by gfwilliams