espruino/BangleApps #935: Bangle.js 2: Not getting time field in GPS fixes

andrewmk, 2021-11-23:
I'm trying to use the gpsrec app on my Bangle.js 2 and it wasn't recording any tracks even though it claimed to have a fix (the widget changed to showing the green flashing lines) and opening the OpenStreetMap app showed a fix and had my location correct. I've just run the gpsrec app while connected to the web ide and I'm getting the following error in the console:
```
Uncaught Error: Cannot read property 'getTime' of undefined
 at line 1 col 699 in gpsrec.wid.js
...;if(hasFix){var period=fix.time.getTime()-lastFixTime;if(period+5...
                                   ^
in function called from system
```

so I tried running this at the console while I have a fix:
```
>Bangle.setGPSPower(1)
=true
>Bangle.getGPSFix()
```

with the following result (I've manually censored part of the coords):
```
>Bangle.getGPSFix()
={ lat: 52.xxxx, lon: -0.7xxxx, alt: 102.6, speed: 0,
  course: 0, time: undefined, satellites: 7, fix: 1, hdop: 1.5 }
```

So it looks like the time field isn't being set.

I'm running firmware version 2v10.222

andrewmk, 2021-11-23:
I've just updated to firmware 2v10.226 and tried again and it seems to be working now. Could just be coincidence of course.

andrewmk, 2021-11-23:
I've just realised this probably should have been on the Espruino repository as it turned out to be a problem with the underlying code (maybe). But I'm not going to move it now as it seems to be fixed.

gfwilliams, 2021-11-24:
Hi, thanks! I think this was the right place for it.

So what it seems happens is the Bangle.js 2 GPS seems to be able to provide a fix without any time, at least for a few readings. I've just made gpsrec cope with that but I guess maybe Bangle.js should ensure fix is 0 even if the time isn't set?

closed 2021-11-24 by gfwilliams

andrewmk, 2021-11-24:
Being a curious sort of person I reverted to firmware 2v10.222 and this time I'm seeing correct behaviour. So I'd just concluded that it probably wasn't the firmware after all and that was indeed a coincidence. Do we want a fix with no time? The OpenStMap app is quite happy with it as it doesn't use the time field. So some apps would work and others wouldn't. Personally I think I'd leave it providing a fix and document that the time field might be undefined, hopefully for a short while. Then the app can decide to treat it as a fix or no fix. 

Just as an aside I think it sometimes seems to provide the time field before there's a proper fix. So yesterday I was seeing no fixes with all the fields null, no fixes with the time filled in and nothing else, fixes with no time and fixes with time.

Frankly I'm at a loss to know how a GPS module can get a fix without knowing precisely what the time is - I thought that was the whole point of how they worked.

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v10.222 (c) 2021 G.Williams
>Bangle.getGPSFix()
={ lat: NaN, lon: NaN, alt: NaN, speed: NaN,
  course: NaN,
  time: Date: Wed Nov 24 2021 12:01:15 GMT+0000,
  satellites: 0, fix: 0, hdop: 25.5 }
>Bangle.getGPSFix()
={ lat: 52.xxxx, lon: -0.7xxxx, alt: 90, speed: 0,
  course: 0,
  time: Date: Wed Nov 24 2021 12:02:26 GMT+0000,
  satellites: 4, fix: 1, hdop: 7.7 }
>
```

commit cd1daba0 (2021-11-24): 0.25: Fix issue where if Bangle.js 2 got a GPS fix but no reported time, errors could be caused by the widget (fix #935)