espruino/BangleApps #941: authentiwatch pc tool not sending tokens to watch

Izaic, 2021-11-24:
When I click the button to upload the tokens to the watch, it shows a loading bar on the bottom for a split second and then nothing happens.

m-p-3, 2021-11-24:
Same here when pairing through mobile (Android 11, using Bromite 95.0.4638.99 as the web browser), the app still says *No tokens* after sending to watch.

Smooklu, 2021-11-27:
Same issue here.

andrewgoz, 2021-11-30:
I'm the author. Until I get an actual Bangle (it's on the way) I can't test the loading/saving of tokens. Other than waiting until I can look at it, another developer might be able to tell me what I'm doing wrong?

Smooklu, 2021-11-30:
> I'm the author. Until I get an actual Bangle (it's on the way) I can't test the loading/saving of tokens. Other than waiting until I can look at it, another developer might be able to tell me what I'm doing wrong?

I can submit the error message I get from the app.

Smooklu, 2021-11-30:
```
Uncaught TypeError: If specifying an object, it must be of the form {data : ..., count : N} or {callback : fn} - got {
  "algorithm": "SHA1",
  "digits": 6, "period": 30,
  "issuer": "",
  "account": "",
  "secret": "JBSWY3DPEHPK3PXP",
  "label": "token"
 }
 at line 1 col 161
...EHPK3PXP","label":"token"}])
```

Smooklu, 2021-11-30:
It looks like the issue is from `interface.html` and not `app.js`.

andrewgoz, 2021-11-30:
Fixed in #972 (checked on my Bangle which turned up while I was at work). It turns out the library Puck.write() has particular structure requirements...

gfwilliams, 2021-11-30:
Ahh - it might actually be that you needed `Storage.writeJSON` and not just `.write` since if you give that an array it assumed it's binary?

m-p-3, 2021-12-04:
Just FYI, I tried pushing a token today to my Bangle.js2 and it worked 🎉

Smooklu, 2021-12-04:
It worked for me too! but I am having a another issue now.

Smooklu, 2021-12-04:
[image]

Smooklu, 2021-12-04:
When I open the app, I see "Loading..." and then a blank screen with widgets loaded.

Smooklu, 2021-12-04:
When I save a token to the watch, it reports `undefined` in the console. Possibly a issue?

Smooklu, 2021-12-04:
```
{"tokens":[{"algorithm":"SHA1","digits":6,"period":30,"issuer":"","account":"","secret":"JBSWY3DPEHPK3PXP","label":"TEST"}],"misc":{}}
```

Smooklu, 2021-12-04:
If I try installing the app without saving a token, it says `No tokens`  which is expected.

andrewgoz, 2021-12-04:
"No tokens" is the only text where I don't explicitly set the text colours. For everything else I use the defined theme colours. Try changing the theme and see if the text appears. You could also try viewing the theme in the IDE:

`>g.theme`
`={ fg: 0, bg: 65535, fg2: 31, bg2: 2047,`
`  fgH: 0, bgH: 2047, dark: false }`

Could you also try the following in the IDE console:

`>require("Storage").writeJSON`
`=function () { [native code] }`

andrewgoz, 2021-12-16:
Just a note that in #981 I switched to the recommended `Storage.readJSON` and `Storage.writeJSON`. I now consider this issue closed. Whatever is happening with @Smooklu is a separate issue.

closed 2021-12-16 by Izaic