espruino/BangleApps #944: Touch Launcher does not show some icons right on low resolution and high resolution it flickers when animating.[Bangle 1]

Ronin0000, 2021-11-24:
The touch launcher app does not show some icons right on low resolution. For example the settings icon looks like a circle with a weird outline and the school calendar icon is messed up. Also, when you turn on high resolution the icons are fine, but the animation of moving between app icons flashes. I put images below. Sorry for the bad images.
[image]
[image]

gfwilliams, 2021-11-24:
This is https://github.com/espruino/Espruino/issues/2049 - if you update your Bangle.js firmware to cutting edge (or 2v11 when it's released soon) it'll be fixed

closed 2021-11-24 by gfwilliams

Ronin0000, 2021-11-24:
ok! Thanks!