espruino/BangleApps #946: Anton and CLI-Clock JS2 clocks display "0:00" every morning.

pzo, 2021-11-25:
Every morning the Anton clock displays "0:00" 
Same with CLI-Clock JS2, the flashing cursor is not displayed.
With both the date is fine, only the time is "frozen".
After going to settings and back time is updated again.

gfwilliams, 2021-11-25:
Do you have any other apps/widgets installed?

Because I don't see that here with the Anton clock. It almost feels like at midnight something is calling 'clearInterval()' which would kill all active timers on the watch

pzo, 2021-11-25:
Yes, I have both the "Pedometer widget" and "GPS widget" active.

gfwilliams, 2021-11-25:
Ok, and that's it? I've just set up a clock with the default apps plus pedometer and gps widget and we'll see what happens

pzo, 2021-11-25:
I also have the default "Bluetooth ID", "Bluetooth" and "Battery level" widgets.

profriemke, 2021-11-27:
If you connect with the IDE, does it show a FIFO error?

gfwilliams, 2021-11-29:
Just to confirm, I have a watch set up with all those things installed and I haven't ever seen the clock displaying `0:00` in the morning.

If you connect with the App Loader and go to 'My Apps' do you see any other apps listed at all? Maybe if that's all you have try clicking the 'Install Default Apps' button in 'About' in the App Loader to reset it to a 'factory' state and see if that fixes it.

pzo, 2021-11-29:
It is now fixed after doing a 'Install Default Apps'. 
Thank you !

pzo, 2021-12-02:
I think I found the culprit after reinstalling QAlarm the problem reappeared.

qalarmcheck.js does : 

clearInterval();
...
And re-executes itself via : 

if (nextAlarms[0]) {                                                                                                                                                                                               
  setTimeout(() => {                                                                                                                                                                                               
    eval(require("Storage").read("qalarmcheck.js"));                                                                                                                                                               
    load("qalarm.js");                                                                                                                                                                                             
  }, nextAlarms[0].t - t);                                                                                                                                                                                         
} else {                                                                                                                                                                                                           
  // No alarms found: will re-check at midnight                                                                                                                                                                    
  setTimeout(() => {                                                                                                                                                                                               
    eval(require("Storage").read("qalarmcheck.js"));                                                                                                                                                               
  }, 86400000 - t);                                                                                                                                                                                                
}                   

Also shouldn't it be clearTimeout/setTimeout  or clearInterval/setInterval ?
It seems they are technically interchangeable but for clarity it would be better to use them together.

gfwilliams, 2021-12-02:
Great, thanks for that! That sounds like it then! Just fixed it with a change on the development app loader.

What is it about the Q Alarm that's better than the normal one out of interest? It feels like there's an awful lot of code copied from the main Alarm app - like maybe the changes should have just been rolled into that instead...

closed 2021-12-02 by gfwilliams

pzo, 2021-12-02:
In Q Alarm being able to configure a day of the week for the alarm is nice.

commit 5b967184 (2021-12-02): 0.03: Fix unfreed memory, and clearInterval that disabled all clocks at midnight (fix #946)
      Fix app icon
      Change menu order so 'back' is at the top