espruino/BangleApps #953: Lock Widget with open lock icon

profriemke, 2021-11-27:
It would be good to have an open lock icon instead of disappearing, which makes the other widgets jump.

gfwilliams, 2021-11-29:
I wonder whether at a glance it would be harder to tell the difference between an open and closed lock? I guess it could keep the width the same and just keep itself blank so nothing moved around though, but personally that's never really upset me

profriemke, 2021-11-30:
I tried another icon. You are right, it's only visible when red and green. A fixed width would be a good idea. I tried this with your code and it looks smooth now. Would be a small, but nice improvement.

mabuchner, 2021-11-30:
> A fixed width would be a good idea. 

With a hidden lock widget, which still takes up space, I would be wondering why my widgets aren't aligned correctly to the left side.

How about just moving the lock widget to the bottom where it is less crowded?

[image]

[image]

Or make it possible to horizontally center widgets and place the lock widget there?

profriemke, 2021-11-30:
This could interfere with other apps:

[image]

btw: the adapted fixed width lock is in the upper left (unlocked).

thyttan, 2022-11-29:
Candidate for being closed - leave a comment if you object!

Rationale:
Light Switch Widget implements the feature request to a large degree if not completely.

gfwilliams, 2022-11-30:
If someone cared enough then they could copy the existing lock widget and change it in under 5 minutes.

But just to avoid complaints, I just added `widlockunlock` widget that does exactly this

closed 2022-11-30 by gfwilliams