espruino/BangleApps #954: Timezone ignored when setting time in IDE or App Loader

profriemke, 2021-11-27:
Timezone is ignored in Bangle.js v2 when setting time in IDE or app loader. Within +1 timezone the clock is always set to +0 time.

gfwilliams, 2021-11-29:
That's an odd one. What does `(new Date()).getTimezoneOffset()` report if you run it in your browser?

The App Loader does definitely set the timezone according to the browser: https://github.com/espruino/EspruinoAppLoaderCore/blob/50aa45f13f06cc2f40684971d4c0a68d061b1f3c/js/comms.js#L292

Although I don't believe the IDE does, so maybe that's what is causing problems?

profriemke, 2021-11-30:
It returns the correct -60.

profriemke, 2021-11-30:
It looks ok. Yes, it could also be the IDE since I switch often between them these days.
At the moment I cannot reproduce it, so I think it's good to close.

closed 2021-11-30 by profriemke

gfwilliams, 2021-11-30:
Ok, thanks! If you can reproduce it please let me know though