espruino/BangleApps #958: TimeCal wong weekday

Philsan69, 2021-11-28:
Today, Sunday 28th November 2021 17:00, appears under the first column (Monday).
Bangle 2, Locale settings Time Zone=1 and Clock style 24hr.
BTW, the app has no icon in the menu.
Thanks to the coder!

gfwilliams, 2021-11-29:
@madman2021  any ideas?

Philsan69, 2021-11-30:
Perhaps the fact Bangle2 isn't connected?
Today I found Anton Clock stuck with 0.00 time.
When I connected time has been updated.
Btw, that's no good because watchfaces should work without connection.

gfwilliams, 2021-11-30:
> Btw, that's no good because watchfaces should work without connection.

They do. I'm not quite sure what's happened in your case - but I imagine you're not using the default apps? Maybe start a post for this in the forum and let us know exactly what apps you have installed http://forum.espruino.com/microcosms/1424/

gfwilliams, 2022-10-28:
I believe this is now fixed with an updated timecal

closed 2022-10-28 by gfwilliams