espruino/BangleApps #961: Pedometer widget step count randomly resetting

midnight4577, 2021-11-29:
Recently while useing the Pedometer Widget I noticted that the step count seems to reset to 0. I don't know why this is happening but my guess is it is either in the way the step count is stored or how the app writes to it.

gfwilliams, 2021-11-29:
That's an odd one - it resets at midnight, but that should be it. I haven't seen any cases where it resets itself before.

If you find something in particular you're doing that makes it reset, please can you let me know?

midnight4577, 2021-11-29:
I was just messing around in the menus apps etc. and it seems to only reset once it updates the step count so it must be something strange in that specific piece of code and I can regularly get it to reset by opening a app that does not load widgets such as Custom QR code reseting back to the watch face and then taking steps to update the count.

Philsan69, 2021-11-29:
I noticed that sometimes digits (Large Digits setting) are overwritten by QR Alarm app icon.

gfwilliams, 2021-11-30:
> I noticed that sometimes digits (Large Digits setting) are overwritten by QR Alarm app icon.

Is that because there isn't space left in the widget bar, or does there just not seem to be enough room? Chances are the Q alarm app has its width set wrong.

@midnight4577 thanks - that's really helpful. Should be quite easy to track down knowing it can be reproduced if an app doesn't load widgets

Philsan69, 2021-11-30:
Pedometer widget is the first one, on the left there's only the lock key icon and on the right QR Alarm bell icon and justified to the right battery and Bluetooth widget icons.

CountMurphy, 2021-12-03:
Having the same issue here. Last night my bangle only recorded 22 steps for the whole day (because of resets). It appears to reset after opening other apps. Currently used apps: gadgetbridge, calculator, health tracking, lcars, and music controls.

Saduff, 2021-12-03:
Have the same issue, step count reset twice today. Didn't open any apps during that time, only got some notifications from Gadgetbridge.

I noticed some kind of blue color behind the step count widget though. And later when it had reset, the blue color was gone.

midnight4577, 2021-12-04:
The blue you are seeing is most likely part of a circle which shows your progress to your goal. If you want to get rid of it simply go to settings, App/Widget settings, Pedometer widget, and turn "Show progress" to off.

hughbarney, 2021-12-05:
I have seen the step count reset at odd times using the pedominter widget.  Not yet got any pattern to reproduce.  Its as if it defaulted to zero and could not read its onkill count back on start up.

gfwilliams, 2021-12-06:
I should add that although the step count widget resets, the actual step count logging from the `health` app won't - so you'll still have good step reporting there.

midnight4577, 2021-12-06:
Sounds like a decent workaround could be to grab the data from the health app although that depends on the app being installed.

hughbarney, 2021-12-07:
Is there a simple call to the health app to get todays step count ?

gfwilliams, 2021-12-08:
Well, you can just ask it for the health info for a day and then sum up steps

gfwilliams, 2021-12-14:
I can't reproduce this. Step counting works absolutely fine for me here and I can't get a reset.

I *think* the issue was to do with the flash writes failing periodically, which was fixed in a firmware update a few weeks ago. I'd be really interested to see it anyone can reproduce with up to date firmware.

CountMurphy, 2021-12-14:
Is the new firmware no longer beta? The change log indicates it is not, but the description still says beta.

gfwilliams, 2021-12-14:
Where do you see the description?

CountMurphy, 2021-12-14:
on banglejs.com/apps. [image]

gfwilliams, 2021-12-14:
Ahh, ok - the firmware isn't beta, it's the firmware updater on the website that is...

Instructions at https://www.espruino.com/Bangle.js2#firmware-updates

CountMurphy, 2021-12-14:
Ah! I'll upload today and let you know if the resets stop. Thanks!

CountMurphy, 2021-12-14:
Running 10.247, step counter reset again on widget.

Saduff, 2021-12-14:
I haven't noticed a reset after upgrading to firmware 239, but I haven't paid that much attention to it lately.

gfwilliams, 2021-12-15:
@CountMurphy ok, thanks. Do you have any steps for getting it to happen reliably?

CountMurphy, 2021-12-15:
I wish I did. I have not been able to notice any pattern. Since you can't reproduce the error, I could ship you my bangle if that would help (as long as I got it back).

WizardStan, 2021-12-15:
I'm also experiencing this.  I thought it was related to bluetooth connection but that was coincidence because I was also trying to debug notification failures at the same time.
I may try 239 later, but the good news (tm) is that I can consistently reproduce the issue on 238.  15 times in a row, at least, with no "failures" (for a certain definition of failure) following these exact steps.
First I walk around a little bit to rack up 30 steps or so.  Stop! Be very careful not make any step-like motions.  Open the menu (press the button twice, once to unlock, once to open the menu), open an app, then close the app.  Now take another few steps.  Counter widget resets back to 1.  I've done this with Messages, Android, Settings, About, and Compass.
If, between opening and closing the app you register a "step" then the counter will go up and the reset won't happen, it has to be completely still.

Receiving a notification opens the Messages app automatically, but closing it does not trigger the issue, nor does opening it from the widget; only by opening the Messages app (or any other app) through the app launcher and then closing it do I see the problem.  There's no time window that I can tell either: open and then immediately close the app, or open it and wait 60 seconds, doesn't matter, step counter is reset.

I still don't entirely understand when and how on('kill') callbacks are called so I could be way off base here, but is it possible it's getting called twice somehow, when opening the app launcher and then when closing the selected app?  And the stp_today is somehow getting reset which results in the second kill saving a 0?  But if you take a few steps, stp_today gets updated and thus you don't see the problem?  Just a thought.

Saduff, 2021-12-15:
> I haven't noticed a reset after upgrading to firmware 239, but I haven't paid that much attention to it lately.

I was wrong, it does reset and I can reproduce it with the steps outlined by @WizardStan.

hughbarney, 2021-12-15:
Well done in reproducing that should narrow things down a bit.

nebbishhacker, 2021-12-16:
I believe I've found the bug.

The date widget saves the time of the last step as UTC timestamp using `lastUpdate.toISOString()`. The time is then loaded using `new Date(pedomData.lastUpdate)`, which apparently interprets the timestamp as local time. Each time the widget is saved and loaded without detecting any steps, the error compounds. When the timestamp gets shifted enough the widget ends up concluding that the last step was was taken on a different day, and it resets the count.

gfwilliams, 2021-12-16:
Brilliant - thanks again @nebbishhacker! It explains why I was finding it so hard to reproduce but it happened really often for some others :)

closed 2021-12-16 by gfwilliams

WizardStan, 2021-12-16:
That tracks.  Explains why sometimes notifications would trigger a reset and sometimes they wouldn't, and why I had no trouble reproducing it at 10pm but took forever to find a consistent way to reproduce it at 10am.  Good job.

PR #1101 (2021-12-16): widpedom: Save lastUpdated as a unix timestamp
Works around espruino/Espruino#2116 in order to fix #961
commit 108c12d3 (2021-12-16): widpedom: Save lastUpdated as a unix timestamp

Fixes #961