espruino/BangleApps #963: Going back from settings should go to app launcher

mabuchner, 2021-11-29:
Whenever I go back from the settings, I have the feeling, that something went wrong, because I'm immediately back at the clock screen.

I'm opening the settings through the launcher

Clock -> Launcher -> Settings

Hence, going back from the settings should open the launcher.

gfwilliams, 2021-12-02:
It's a point, but all apps go back to the clock... I don't think that settings in particular should be the only one to do something different...

gfwilliams, 2022-03-01:
I think so too - thanks for the backup :)

Just as an aside, if someone wanted it'd be relatively easy to create an app that override `E.showMenu` and detected any `< Back` entries that contained a call to `load()`, and then changed it to load the launcher instead.

closed 2022-03-01 by gfwilliams