espruino/BangleApps #971: Closing app launcher with short button press

mabuchner, 2021-11-30:
Right now it feels a bit unnatural for me to close the app launcher on my Bangle.js 2 via a button press.

I would expect, that a short button press is sufficient to go back, but one has to deliberately hold down until the loading message appears.

It would be nice, if a short button press would be sufficient.

gfwilliams, 2021-11-30:
Yes - it's been suggested that on other devices with a single button, the button is generally used to go Back and the touchscreen used to go forward - so I think we might move in that direction

mabuchner, 2021-11-30:
Wouldn't adding the following code (which I found int he About app) in the [launcher](https://github.com/espruino/BangleApps/tree/master/apps/launch) already be sufficient?

```js
setWatch(_=>load(), BTN1);
```

gfwilliams, 2021-11-30:
Yes, that should work just fine

commit 9d26a644 (2021-12-02): launcher 0.09: Bangle.js 2 - pressing the button goes back to clock (fix #971)
      After 10s of being locked, the launcher goes back to the clock screen
closed 2021-12-02 by gfwilliams